pellepl/spiffs #119: Unable to write large File

sensslen, 2016-12-13:
I've been poking around with spiffs for a while now and everything seemed to work properly so far. 

Now I try to integrate spiffs into our product and can't write a large (15516 bytes) file. It's somewhat special that I try to write this amount of data with an offset of 6 bytes.

But when I checked this does indeed work when using a smaller file size.

The Error I get is -10015 which I get because I pass the maximum value of data_page (0xFFFF) in spiffs_page_data_check on line 1324 of spiffs_nucleus.c.

Unfortunately I haven't been able to figure out it's calculation. Some insight into the calculation of this data_page would be much appreciated.

SPIFFS outputs the following on mount:
CACHE_ALLO: allocated cache page 0
CACHE_ALLO: allocated cache page 1
CACHE_ALLO: allocated cache page 2
CACHE_ALLO: allocated cache page 3
CACHE_FREE: free cache page 0 pix 0000
CACHE_ALLO: allocated cache page 0
CACHE_FREE: free cache page 1 pix 0001
CACHE_ALLO: allocated cache page 1
CACHE_FREE: free cache page 2 pix 0002
CACHE_ALLO: allocated cache page 2
CACHE_FREE: free cache page 3 pix 0003
CACHE_ALLO: allocated cache page 3
CACHE_FREE: free cache page 0 pix 0100
CACHE_ALLO: allocated cache page 0
CACHE_FREE: free cache page 1 pix 0101
CACHE_ALLO: allocated cache page 1
CACHE_FREE: free cache page 2 pix 0102
CACHE_ALLO: allocated cache page 2
CACHE_FREE: free cache page 3 pix 0103
CACHE_ALLO: allocated cache page 3
CACHE_FREE: free cache page 0 pix 0200
CACHE_ALLO: allocated cache page 0
CACHE_FREE: free cache page 1 pix 0201
CACHE_ALLO: allocated cache page 1
CACHE_FREE: free cache page 2 pix 0202
CACHE_ALLO: allocated cache page 2
CACHE_FREE: free cache page 3 pix 0203
CACHE_ALLO: allocated cache page 3
CACHE_FREE: free cache page 0 pix 0300
CACHE_ALLO: allocated cache page 0
CACHE_FREE: free cache page 1 pix 0301
CACHE_ALLO: allocated cache page 1
CACHE_FREE: free cache page 2 pix 0302
CACHE_ALLO: allocated cache page 2
CACHE_FREE: free cache page 3 pix 0303
CACHE_ALLO: allocated cache page 3
CACHE_FREE: free cache page 0 pix 0400
CACHE_ALLO: allocated cache page 0
CACHE_FREE: free cache page 1 pix 0401
CACHE_ALLO: allocated cache page 1
CACHE_FREE: free cache page 2 pix 0402
CACHE_ALLO: allocated cache page 2
CACHE_FREE: free cache page 3 pix 0403
CACHE_ALLO: allocated cache page 3
CACHE_FREE: free cache page 0 pix 0500
CACHE_ALLO: allocated cache page 0
CACHE_FREE: free cache page 1 pix 0501
CACHE_ALLO: allocated cache page 1
CACHE_FREE: free cache page 2 pix 0502
CACHE_ALLO: allocated cache page 2
CACHE_FREE: free cache page 3 pix 0503
CACHE_ALLO: allocated cache page 3
CACHE_FREE: free cache page 0 pix 0600
CACHE_ALLO: allocated cache page 0
CACHE_FREE: free cache page 1 pix 0601
CACHE_ALLO: allocated cache page 1
CACHE_FREE: free cache page 2 pix 0602
CACHE_ALLO: allocated cache page 2
CACHE_FREE: free cache page 3 pix 0603
[... 989 lines cut ...]
CACHE_GET: have cache page 1 for 0004
read: offset:20 rd:12 data spix:0000 is data_pix:0005 addr:0000051d
CACHE_GET: have cache page 2 for 0005
CACHE_GET: have cache page 2 for 0005
read: load objix page 0004:0000 for data spix:0000
CACHE_GET: have cache page 1 for 0004
read: offset:32 rd:12 data spix:0000 is data_pix:0005 addr:00000529
CACHE_GET: have cache page 2 for 0005
CACHE_GET: have cache page 2 for 0005
CACHE_GET: have cache page 0 for 0000
CACHE_GET: have cache page 1 for 0004
CACHE_FREE: free cache page 3 pix 7f03
CACHE_ALLO: allocated cache page 3
CACHE_GET: have cache page 0 for 0000
open: fd 2 is obj id 80000040
CACHE_GET: have cache page 0 for 0000
CACHE_GET: have cache page 3 for 0006
CACHE_GET: have cache page 3 for 0006
CACHE_GET: have cache page 0 for 0000
open: fd 2 is obj id 80000040
append: 15516 bytes @ offs 6 of size -1
append: 80000040 load objixhdr page 0006:0000
CACHE_GET: have cache page 3 for 0006

This resembles reading a File called "00" which contains data on the available files and then trying to write 15516 Bytes.

pellepl, 2016-12-13:
I cannot find which spiffs_page_data_check you refer to, probably because you're running an older version. From the logs it looks like you create an empty file, then moves the offset to 6, and finally writes the 15516 bytes. spiffs cannot handle [offset moving beyond file end](https://github.com/pellepl/spiffs/wiki/Using-spiffs#seeking-in-a-file) so you should have got an error from the lseek in the first place. -10015 is a fatal error and indicates a broken file structure, i.e. "it should not happen" :)

sensslen, 2016-12-14:
Ok, this is interresting. I'm usin version 3.6 which is the last released version of SPIFFS.

I thought that seeking over the file's end might be the problem so I created a Test where I seeked to position 6 in a newly created File and wrote 10 Bytes. This worked.

Obviously seek is not failing but the write command is.

The failing function call resides on line 1312 in file spiffs_nucleus.c in function spiffs_object_append.

I'll try using spiffs without seeking over the end of the file and see whether it works like this.

pellepl, 2016-12-14:
Ah, you've found a bug! The seek function returns error when seeking beyond end, which is expected. But for for virgin files that have never been written to, seek just passes through and skips all verification.

There are problems having a fully posix-compliant seek in spiffs wrt the nature of spi flashes and the spiffs structure. If one really wants gaps in files, as in your case, I'd recommend writing six bytes worth of 0xFF instead of seeking. Adding gaps will put a lot of extra complexity in the code.

Thanks for finding this! I'll add a report for this and fix it when I find the time.

Cheers / Peter

closed 2016-12-14 by pellepl