pellepl/spiffs #123: Question: files order

valkuc, 2016-12-15:
Hello. Does Spiffs preserve file creation order on dir list operations? 
I need to make a file based queue with limited size. When queue size reaches N records, I need to delete first created file and add new. So the order of insertion is critical in this case.
I have noticed that after some time of usage file order getting wrong. I.e. listing directory not return first created record but some other. Am I doing something wrong or SPIFFS just not guarantee that order of files when listing will be same as order of creation of those files?

pellepl, 2016-12-15:
Hi,
Once the garbage collection kicks in the file order will be scrambled. You need to keep the ordering info yourself, for instance in the filename or in the actual file.

valkuc, 2016-12-15:
Thanks. I guess it will be good to add this info in README.md

closed 2016-12-15 by valkuc

pellepl, 2016-12-18:
FYI, added to the FAQ in the wiki: https://github.com/pellepl/spiffs/wiki/FAQ#will-file-list-order-be-preserved

Cheers / P