pellepl/spiffs #125: SPIFFS_rename behaviour

rojer, 2017-01-09:
I don't expect this to be addressed immediately, but perhaps in some future revision...

`rename()` is commonly used to safely write files, i.e. prepare a temp file and then move it into place. it's a good practice, but its safety hinges on the atomicity of rename and its behavior wrt destinations that already exist. it would be nice if SPIFFS_rename behaved closer to POSIX, specifically: it should be atomic and should not fail if `newPath` exists, it should silently delete `newPath` in this case.

see [here](http://pubs.opengroup.org/onlinepubs/9699919799/functions/rename.html) for detailed spec.

pellepl, 2017-01-09:
Ah, I did not know that this was the posix way. Will fix in som indefinable
future 😉. Thanks for the heads up!

Cheers / P

rojer, 2017-01-09:
perhaps the latter (delete if `src != dst` and `dst` already exists) is simpler to implement.