pellepl/spiffs #126: mount / unmount / mount issue?

robynjayqueerie, 2017-01-11:
I tried writing code which basically mounts the flash writes some files then unmounts and loops back again to the mount.
All seems to go OK for a while then I start to get read and write errors (say after 3 cycles)
If I mount once and never unmount then the writes continue as expected overwriting the originals.
I have checked my erase functions, chip is fully erased and Spiffs does not call any erase function before the errors show up.
Should I expect to be able to do a mount/unmount and still have it function OK? It would seem the thing to do between usage of the flash which may be days between writes in my case.
Thanks

pellepl, 2017-01-11:
Hi,
Yes that should definitely work. What errors do you get?

robynjayqueerie, 2017-01-11:
Hi,

The errors are not consistent each time although they tend to follow a pattern.

The folder I am using for the project is copied to here spiffs (file://ROBYN-PC/Users/Robyn/Dropbox/Public/spiffs <file:///\\ROBYN-PC\Users\Robyn\Dropbox\Public\spiffs> ) 

The program main.ino is set up to run in the loop() mounting and unmounting. Apart from that and my interface to my Flash routines it is your example program.

I am using a 512Kbyte Flash and set Spiffs up to work on a page size of 256 bytes and a block size of 4K at the moment. The flash routines have been changed to check after erase. It would be interesting if you could run the program with your flash routines to see if you got a similar effect. 

If I just put the mount in the setup() section and remove the mount/unmount from loop() then it works fine.

Possibility is that my unmount() routine is not correct but it seems straightforward.

Thanks

Robyn

 

Robyn Hannah PhD BE (Hons 1) Grad Dip Counselling.

 <http://nrie.com.au/NRIEWB9Version/index.html> NRIE Logo

NRI Electronics Pty. Ltd.

PO Box 20, Alstonville, NSW, 2477.

Phone: 02 6628 7961 or 0427 068 257

email:  <mailto:contact@nrie.com.au> mailto:contact@nrie.com.auWeb www.nrie.com.au

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

robynjayqueerie, 2017-01-11:
Peter,

I ran it again here as mount/unmount and captured the attached logout.

I had changed the logical page size to 512 from the link I had given you earlier but that is the only change.

Regards

Robyn

 

 

Robyn Hannah PhD BE (Hons 1) Grad Dip Counselling.

 <http://nrie.com.au/NRIEWB9Version/index.html> NRIE Logo

NRI Electronics Pty. Ltd.

PO Box 20, Alstonville, NSW, 2477.

Phone: 02 6628 7961 or 0427 068 257

email:  <mailto:contact@nrie.com.au> mailto:contact@nrie.com.auWeb www.nrie.com.au

 

https://www.avast.com/antivirus

testing the 64K block for erase
mount res: 0
--> Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
¡¤´â  <--

mount res: 0
--> Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
[... 581 lines cut ...]
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hello world this is your master 
Hell <--

pellepl, 2017-01-13:
Hi Robyn,

I cannot see any links I'm afraid. I did however try a thing in the testbench:
First, I created an fs worth of 1 mbyte, with 4k blocks and 512 byte pages.
Then I looped this sequence a hundred times:
1) mount
2) list all files
3) create and write four files (named file1, file2, file3, file4) with different sizes (100, 500, 1000, 2000 bytes), overwriting if already there
4) read them back and verify contents
5) unmount

For what it's worth, it worked in the testbench. Is there any chance the ram becomes corrupt between your unmount and mount?

robynjayqueerie, 2017-01-13:
Evidently replying direct to the emails does not copy the links.
The dropbox link to my folder is \\ROBYN-PC\Users\Robyn\Dropbox\Public\spiffs 
The capture file i obtained from the main.ino routine is
[capture.txt](https://github.com/pellepl/spiffs/files/705547/capture.txt)
I don't think the ram is being corrupted, all I have between the unmount and the mount is a 10 second delay

pellepl, 2017-01-14:
Hi, sorry, but the Dropbox link still just seem to be plaintext. The log
works though.

Den 14 jan. 2017 12:23 AM skrev "robynjayqueerie" <notifications@github.com

robynjayqueerie, 2017-01-16:
If all else fails resort to a zip file. See if this makes some sense.
[spiffs.zip](https://github.com/pellepl/spiffs/files/707338/spiffs.zip)
Thanks for your help
Robyn

pellepl, 2017-01-16:
Ah, good old reliable zip :)

Well, at a first glance (I'm no arduino guy though), your main loop looks good. However, I did find something odd in spiffs/FLASH/flash.cpp. When you write and erase, you check the SR for completion of the write operation. What you do is pulling chip select, sending 0x05 command (read sr), send 0x00 to read the sr, and whilst bit one of the read is high you continuously send 0x00 to re-read the sr. I think the problem is that the subsequent reads will contain results from command 0x00 instead.

Instead, try doing this when checking if spi flash is busy:
``` C
  d = 0xff;
  while (d & 0x01) {
    digitalWrite(flash_CS, LOW);
    SPI.transfer(0x05);
    d = SPI.transfer(0x00);
    digitalWrite(flash_CS, HIGH);
  }
```

robynjayqueerie, 2017-01-16:
Thanks Peter,

You could be right, I will give it a go here and see what is different. The data sheet suggests that I just set the initial address and then clock in to see when it changes but it is a bit ambiguous.

I have been able to set up Spiffs to work with the flash, I am downloading firmware files via ftp, checking their md5 count, saving them with spiffs and then recalling to reflash the device’s firmware. However I have to only mount once or I screw it up.

I will give you the feedback

Robyn

 

 

Robyn Hannah PhD BE (Hons 1) Grad Dip Counselling.

 <http://nrie.com.au/NRIEWB9Version/index.html> NRIE Logo

NRI Electronics Pty. Ltd.

PO Box 20, Alstonville, NSW, 2477.

Phone: 02 6628 7961 or 0427 068 257

email:  <mailto:contact@nrie.com.au> mailto:contact@nrie.com.auWeb www.nrie.com.au

 

https://www.avast.com/antivirus

pellepl, 2017-01-16:
> auth/AFEuidjG_ZYAh1DKaK6rm8GMItxOSw5tks5rSx6PgaJpZM4Lgak5> . <

Hi Robyn,

Yes when I think of it that is the standard way on spi flashes. However, on
some flashes there are more than one sr-byte, so it may be worth giving it
a shot anyway. You might have the case of reading sr1 then sr2 then sr1 ..
etc. I guess this would mostly be a problem when you start erasing as you
would not wait long enough.

But it is very weird that it works with one mount only. Practically, there
is not much reason to unmount actually, as unmounting equals closing all
filedescriptors. But it should work.

Perhaps the config structure has changed somehow between the first and
subsequent mounts. May be worth checking also.

robynjayqueerie, 2017-01-16:
Peter,
yes I tried the change to the status but it does not make any difference. The new capture.txt file is attached
[capture.txt](https://github.com/pellepl/spiffs/files/709617/capture.txt)
It does seem to work for 3 or 4 mount/unmount sequences but then starts to fail, recovers at times and fails again.
Is there a fuller explanation of the error codes available? Apart from a general idea of what they are reporting I really don't have much concept of where to start.
Thanks for you help, the platform is a Particle Electron board on a custom built shield that we have developed. It is based on Arduino initially, shares similar software
[https://community.particle.io/t/anybody-using-a-particle-device-with-sdi-12-interface/22649/24](url)

pellepl, 2017-01-18:
Hi Robyn,
unfortunately no, there is no fuller explanation of the errors other than the code itself. Looking at the capture.txt, the first failing run gives -10004 (ERR_DELETED) at open for write. This do not make much sense as you have CREAT|TRUNC|RDWR as flags. 
I think we need to take it step by step. First, let us check that the spi flash is ok before each mount. In your FLASH/flash.cpp, please add something like this:
```C
uint32_t flashReadJEDEC(void) {
  uint32_t id = 0;
  digitalWrite(flash_CS,LOW);
  SPI.transfer(0x9f);
  id = (id<<8) | SPI.transfer(0xff);
  id = (id<<8) | SPI.transfer(0xff);
  id = (id<<8) | SPI.transfer(0xff);
  digitalWrite(flash_CS,HIGH);
  return id;
}
```
and then in your `loop` in main.ino, add
```C
Serial.printf("jedec id: 0x%08x\n", flashReadJEDEC());
```
before the call to mount.

If the spiflash/communication is ok, you'll get the same id number in every loop.

You can also try to add this in the loop, after your mount:
```C
my_spiffs_mount();
Serial.printf("AFTER MOUNT\n");
SPIFFS_vis(&fs);
```
and this before the unmount
```C
Serial.printlnf("--> %s <--\n", buf);	
Serial.printf("BEFORE UNMOUNT\n");
SPIFFS_vis(&fs);
my_spiffs_unmount();
```

Above needs you to redefine spiffs_config a bit, setting 
```C
#define SPIFFS_TEST_VISUALISATION         1
```
and
```C
#define spiffs_printf(...)                Serial.printlnf(__VA_ARGS__)
```
This will give us a shallow dump of the filesystem before and after modification.

Finally, to behave correctly, at each error we should try to close the fd and unmount directly.  A summarized alteration of the main loop should then look something like this:

```C
void loop(){
char buf[1024];
char cnt;
	Serial.printf("jedec id: 0x%08x\n", flashReadJEDEC());

	my_spiffs_mount();
	Serial.printf("AFTER MOUNT\n");
	SPIFFS_vis(&fs);

	// create a file, delete previous if it already exists, and open it for reading and writing
	spiffs_file fd = SPIFFS_open(&fs, "my_file", SPIFFS_CREAT | SPIFFS_TRUNC | SPIFFS_RDWR, 0);
	if (fd < 0) {
		Serial.printlnf("SPIFFS_open errno %i\n", SPIFFS_errno(&fs));
		goto spiffs_error2;
	}
	// write to it
	for(cnt=0;cnt<30;cnt++){
		if (SPIFFS_write(&fs, fd, (u8_t *)"Hello world this is your master \r\n", 34) < 0) {
			Serial.printlnf("SPIFFS_write errno %i\n", SPIFFS_errno(&fs));
			goto spiffs_error1;
		}
	}
	// close it
	if (SPIFFS_close(&fs, fd) < 0) {
		Serial.printlnf("SPIFFS_close errno %i\n", SPIFFS_errno(&fs));
		goto spiffs_error2;
	}

	// open it
	fd = SPIFFS_open(&fs, "my_file", SPIFFS_RDWR, 0);
	if (fd < 0) {
		Serial.printlnf("SPIFFS_open errno %i\n", SPIFFS_errno(&fs));
		goto spiffs_error2;
	}

	// read it
	if (SPIFFS_read(&fs, fd, (u8_t *)buf, 1023) < 0) {
		Serial.printlnf("SPIFFS_read errno %i\n", SPIFFS_errno(&fs));
		goto spiffs_error1;
	}

spiffs_error1:
	// close it
	if (SPIFFS_close(&fs, fd) < 0) {
		Serial.printlnf("SPIFFS_close errno %i\n", SPIFFS_errno(&fs));
	}
	// check it
	Serial.printlnf("--> %s <--\n", buf);	
spiffs_error2:
	Serial.printf("BEFORE UNMOUNT\n");
	SPIFFS_vis(&fs);
	my_spiffs_unmount();
	delay(10000);
}

```

Due to the nature of the problem, I feel something might be tampering with the ram. Are you running a multithreaded system? Perhaps there is another thread running talking to the flash also? Related to this, I'm wondering if you have any stack guards setup? Spiffs can be pretty stack hungry, and  perhaps the stack top is hit. Though, then you should have the same problem without mounting/unmounting of course.

robynjayqueerie, 2017-01-18:
Thanks Peter, will do, probably Friday, I have to see some clients tomorrow.

Robyn

 

 

Robyn Hannah PhD BE (Hons 1) Grad Dip Counselling.

 <http://nrie.com.au/NRIEWB9Version/index.html> NRIE Logo

NRI Electronics Pty. Ltd.

PO Box 20, Alstonville, NSW, 2477.

Phone: 02 6628 7961 or 0427 068 257

email:  <mailto:contact@nrie.com.au> mailto:contact@nrie.com.auWeb www.nrie.com.au

 

 
 
 
 
 

https://www.avast.com/antivirus

robynjayqueerie, 2017-01-24:
Hi Peter,
Sorry to be a bit longer getting back to you. Life got in the way.
I have implemented the diagnostics you suggest up to providing the shallow dump of the system. I am attaching the capture.txt.
I also had the chips changed on the boards to larger chips, they are now 1MByte chips and I am using all of the space for Spiffs. So there has been that small change to the configuration.
The flashReadJEDEC() result is constant each call jedec id: 0x001f8501
I have not implemented the last part of your email about "close the fd and unmount directly. " I thought I would let you look at the results to this stage first.
In my full project I am not having any problems downloading firmware files and using them to reflash the device, all using Spiffs. But I need to ensure I don't unmount. There are 4 firmware files 2 at about 128KBytes and two a bit smaller.
Thanks for your help.
Robyn
[capture.txt](https://github.com/pellepl/spiffs/files/725449/capture.txt)

pellepl, 2017-01-26:
Hi Robyn,

no problem, life has a tendency to do this to you ;)

Ok, I think we can rule out the spiflash driver. Looking at the dumps something fishy happens at the fourth iteration. I would like to try to reproduce this in the testbench if possible, it does indeed look like a bug.
Could you please post me your spiffs_config.h and the exact contents of how you're mounting, and your loop? I tried running something similar in the testbench before, but it seems it is dependent on details, so I want to make sure I am running the exact same setup as you when it comes to filesystem configuration and usage.

robynjayqueerie, 2017-01-26:
Peter, I have attached the zip file of the whole project as I ran it last and created that capture file. Thanks
[spiffs.zip](https://github.com/pellepl/spiffs/files/733748/spiffs.zip)

pellepl, 2017-01-27:
Hi Robyn,

Well.. I've hammered at it pretty good, trying to reproduce the exact thing. To no avail I'm afraid. However, eventually I did one thing in your spiflash driver. Try commenting out line 217 in flash.cpp, here: 
```C
	SPI.transfer(b);
	SPI.transfer(c);
	while(length > 0){
		*fptr = SPI.transfer(0x00);
		fptr++;
//		*fptr = 0; // EVIL LINE
		length--;
	}
	digitalWrite(flash_CS,HIGH);
```

That line is evilness which make a wild write beyond the area given, which in turn would corrupt the stack. This would be my best bet for the problem as of now at least.

robynjayqueerie, 2017-01-27:
Peter, Brilliantly found, I did change that and it has been cycling happily for 20 minutes.

I think I had that line there because during testing I was saving long strings and reading them back, it was easier to read the text and see if there were errors in it. The NULL at the end was an automatic end of string.

Thank you, I will keep testing but it seems fine.

Robyn

 

 

Robyn Hannah PhD BE (Hons 1) Grad Dip Counselling.

 <http://nrie.com.au/NRIEWB9Version/index.html> NRIE Logo

NRI Electronics Pty. Ltd.

PO Box 20, Alstonville, NSW, 2477.

Phone: 02 6628 7961 or 0427 068 257

email:  <mailto:contact@nrie.com.au> mailto:contact@nrie.com.auWeb www.nrie.com.au

 

https://www.avast.com/antivirus

someburner, 2017-01-27:
Just an observer here.. Your "null terminator" is actually a buffer overflow generator :) 

Take for example, if fptr pointed to an array of size 1 and you wanted to read 1 byte:

```
uint8_t fptr[1];
length = 1
while(length > 0)
{
   *fptr = SPI.transfer(0x00); // fptr[0] - ok
   fptr++; // now points to fptr[1].. oh dear..
   *fptr = 0; // kabooom! well, maybe, who knows what that is.
  length--; //exit loop, cross fingers
}
```
And even if you do account for that with a bunch of checks, you'll still run into trouble because you'll also be doing that to internal spiffs operations. So you'll want to do that outside the flash driver.

Also, I noticed you're using sizeof() to get the length for your reads/writes, at least in the test code. I'm guessing that's what led you to putting in the `*fptr = 0` in the first place- it's writing the entire buffer regardless of actual string length. And you're not 0ing the buffer so the excess elements could be anything. So use `memset(str, 0, sizeof(str))` before strcpy and you should be able to print the string after reading as long as your buffer is big enough.

pellepl, 2017-03-02:
feel free to reopen if necessary

closed 2017-03-02 by pellepl