pellepl/spiffs #127: File metadata

rojer, 2017-01-19:
While thinking of the best way to implement file-level encryption I naturally came up with the need to keep the real file size (as you mentioned in your own comment [here](https://github.com/nodemcu/nodemcu-firmware/issues/1621#issuecomment-273570326)). However, I think that a more generic "file metadata" facility could be useful - ability to attach opaque user data to files. In my case, it would be size. It could also be mtime, for example.
Methods to get/set this metadata should be provided - e.g. `SPIFFS_meta_set/get`.
I suggest that metadata be an arbitrary sized field, size specified at compile time with, say, `SPIFFS_FILE_META_SIZE` which will default to 0. If non-zero, it'll be a simple array of `u8_t`s that is opaque to SPIFFS.

what do you think?

pellepl, 2017-01-19:
Yes, I can see the usefulness in that. I need to ponder a bit over this, I
guess this metadata would be a part of the file header where the other meta
info goes (name, length, etc). It would need some additions to the api I
gather, like SPIFFS_update_metadata. Regarding the mtime idea, it would
also need some kind of callback architecture also (which basically already
is in place). I need to catch up with spiffs, been busy elsewhere.

All in all it is a good idea.

rojer, 2017-01-19:
yes, metadata should go into `spiffs_page_object_ix`, where name and size currently are.
mtime is not a requirement, and can be left to the user to be stored in the metadata if desired. this is what i personally plan to do: have the [VFS layer](https://github.com/cesanta/mongoose-os/blob/7c977a3550bc5ef4018118ba90a0d79d70bf32f8/common/spiffs/spiffs_vfs.c) update mtime on every write as well as perform encryption. so, no need to implement mtime in spiffs itself, as long as we have generic metadata facility.
at the most basic level, we only need set API and `SPIFFS_stat` could be used to get the current md along with other things.

rojer, 2017-01-19:
since i need this right now, if we agree on the API i can do the legwork and send a PR your way.

rojer, 2017-01-19:
Please take a look at the patch above

pellepl, 2017-01-19:
At a quick glance it looks great. Thanks for the test cleanups! I will have a deeper look into it tomorrow. I had a thought on adding a callback when there is a header update (but not from gc), in order to support the time attribute feature. That can be added later on without interfering with what you've done though. 
Again, great work - I'll dig into it tomorrow! Cheers

closed 2017-01-20 by pellepl