pellepl/spiffs #130: Attempt to write out of bounds in the flash

pjsg, 2017-01-24:
The following sequence of operations results in an out-of-bounds write:

```
  open("6file6.xxxxxxxxxxxxxx", 0x1e) -> 4097
  write(4097, , 1014) -> 1014
  write(4097, , 63) -> 63
  remove("6file6.xxxxxxxxxxxxxx")
  close(4097)
  open("6file6.xxxxxxxxxxxxxx", 0x1e) -> 4097
  write(4097, , 1014)          FATAL write addr too high 01fff00c + 00000002 > 00007000
```

This is one of the test cases found by AFL (PR #100). 

This is just open, write, write, remove, close. Then open, write all on the same file.

pellepl, 2017-01-26:
Thanks @pjsg, I'll look into it. I plan to do a complete overhaul of the fuzzing cases you've found so far - I haven't forgot it and really appreciate the effort. I looked at some of them briefly before, one of them is an extreme case of configuration. There was some planning adding extra error codes preventing such configs from the start, but something came in the way.

pellepl, 2017-07-15:
Fixed by 03e4f80

closed 2017-07-15 by pellepl