pellepl/spiffs #133: SPIFFS writing the same address without erase in between

szekelyisz, 2017-02-24:
Hi, while trying to integrate SPIFFS with https://github.com/Marzogh/SPIFlash I found that SPIFFS writes to the same address on the flash for a second time without erasing it first. This pisses off SPIFlash, because it checks before every write if that location has already been written to and allows the write only if it hasn't (it checks if the location contains `0xff` or something else). This kind of behavior is reasonable for a flash memory, but is it compatible with SPIFFS? Or is SPIFFS smart enough to handle this (without an erase)? Or did I overlook something, and this double write shouldn't happen? A very simple test case of creating a new file and writing any data to it triggers this error. Here are my relevant settings:

```
#define SPIFFS_USE_MAGIC                (1)
#define SPIFFS_USE_MAGIC_LENGTH         (1)
#define SPIFFS_SINGLETON 1
#define SPIFFS_CFG_PHYS_SZ(ignore)        (1024*1024*16)
#define SPIFFS_CFG_PHYS_ERASE_SZ(ignore)  (65536)
#define SPIFFS_CFG_PHYS_ADDR(ignore)      (0)
#define SPIFFS_CFG_LOG_PAGE_SZ(ignore)    (256)
#define SPIFFS_CFG_LOG_BLOCK_SZ(ignore)   (65536)
typedef u16_t spiffs_block_ix;
typedef u16_t spiffs_page_ix;
typedef u32_t spiffs_obj_id;
typedef u16_t spiffs_span_ix;
```

I'm using the current master branch. Thanks!

pellepl, 2017-02-24:
Hi there,
this is as implemented - spiffs is designed this way to update flags, forcing certain bits to zero to reflect metadata changes, which in turn will save flash memory. The spiffs - spi flash HAL should merely do what specified - read, write, and erase.

Cheers / P

szekelyisz, 2017-02-24:
Alright, I think it's time to drop SPIFlash and use your spiflash_driver instead. Thanks!

closed 2017-02-24 by szekelyisz