pellepl/spiffs #138: FS Enc/Dec

mardanikamali, 2017-03-27:
Hello,

I integrated your spiffs project with my architecture, and it works well. 
I wanna implement an encryption level (like AES) on the FS. I looked for it in your project, but I did not find anything. So, I wanna develop my own encryption. I have some questions, and I would be grateful if you help me about this concerns.

1. Our target algorithm is AES-XTS, which encrypts the FS page by page, and each encryption operation encrypts 128 bits of the FS. Also, if we had a write operation whose size is not 128*X, we need to use padding. Default value in the FS is "0xFF", So it will be preferred to use "0xFF" for padding, but the encrypted value is not "0xFF". So, our next write operation might be masked by some non-"0xFF" values, which are written in previous write operation. (we have multiple open/write/close on a file)

2. Sometimes caching helps to develop encryption on FS. Caching allows writing just before closing file. so it can aggregate all writes and push it before closing. In the meantime, I disabled CACHE_WR in SPIFFS, but I will be happy if you let me know that what the structure of cache is in SPIFFS.

Do you have any recommendation for encryption in NVMs? If you have any experience in this issue, I would be appreciative if you inform me.

pellepl, 2017-04-02:
Hi  there,

encryption is unfortunately left as an exercise for the user :)

However, there is at least one known attempt on this: https://github.com/nodemcu/nodemcu-firmware/issues/1621

1. Regarding the padding, this poses a problem in spiffs. Say you write 15 bytes, and then append 8 more. From a non encrypted view, the 8 byte write will be an append operation. Here, spiffs will actually append these bytes on an existing page (over the 0xff:s) if they fit. From an encrypted 128-bit view, this will be an update operation, as the first 15 byte write will in reality be a 16 byte write, and the last 8 byte write will actually be a 16 byte overwrite/update and a 16 byte append. In the issue above, the guys solved this in a clever way by adding extra metadata - spiffs keeps track of the actual encrypted file length, whilst the extra metadata keeps track of the unencrypted file length. In my opinion, this is the easiest way to do it.

2. I need to refresh myself on how the caches work in detail, but I am pretty certain these cannot help you. The thing is that the caches work differently for read cache and write cache. Also, there is no way to actually encrypt the metadata of spiffs.

Hope this helps / P

rojer, 2017-04-04:
yes, i implemented file-level encryption as part or [SPIFFS VFS layer](https://github.com/cesanta/mongoose-os/blob/22ee284f604f3603f3514ecdb040d7b59f20a60e/common/spiffs/spiffs_vfs.c) in Mongoose OS: file names and content are encrypted, but not the fs metadata. this is less than ideal, but i don't think there's a feasible way to do it at a lower level without major rewrite.

Note: this work is not part of the SPIFFS project and the open source version is licensed under GPL, like the rest of Mongoose OS.

closed 2017-04-24 by pellepl