pellepl/spiffs #141: Different object ID

spe-ciellt, 2017-04-12:
Hi,

When looping out existing files using opendir()/readdir()/closedir() I get object ID with most significant bit set (0x8000). When I query on a separate file using stat/fstat I get object ID with the most significant bit not set. I see in spiffs_hydrogen.c function spiffs_stat_pix you do:
`s->obj_id = obj_id & ~SPIFFS_OBJ_ID_IX_FLAG;`

Should the masking be made on obj_id reponse from readdir() too?

Or am I out biking? ;)

Thanks,
/Stefan

rojer, 2017-04-12:
this was (or was supposed to have been) fixed in https://github.com/pellepl/spiffs/commit/bc87d0b44cd6e70147572173cd068a0b1cbc8b33

spe-ciellt, 2017-04-12:
Oh, didn't notice, thank you! Seems like I have some patching to do.

closed 2017-04-12 by spe-ciellt