pellepl/spiffs #142: SPIFFS full open remove file problem

Dan9885, 2017-04-21:
Dear all,
I have a problem with the SPIFFS_open and SPIFFS_remove functions. I create a file(temp.dat), then I append some data to it and at last I rename it(log.dat). After this,  I re-create the file with the same name(temp.dat), and I fill the file with the same procedure.
This is done many times until the file system is full . I saw that after the file is deleted the page index (pix) continues from the last used by the file before it was removed. In this way after many times I reach the end of file system. I expected that, after the file removal, the pix comes back, to permit to use the free memory.
I have 1.5MB of space in flash memory used for SPIFFS. If I create and remove the same file many times the spiffs shouldn’t  be full.

For example: I create the file and pix is 2206, then I fill the file and pix goes to 2418. At last I delete the file, then I re-create but pix remains 2418.

comino, 2017-04-21:
I can confirm this behaviour: 

Reproduce via: 
1) Create a file "myfile.dat.temp" and fill with data (im my case I fill it with 50kB)
2) If exists, delete with myfile.dat
3) rename myfile.dat.temp to myfile.dat 
4) Repeat 

SPIFFS is latest github version.

pellepl, 2017-04-23:
@comino, what is the confiugration of your filesystem when it comes to logical page sizes and logical block sizes? I'll try to reproduce it here.

comino, 2017-04-24:
A example to reproduce 
` 
    SPIFFS.begin();

    std::vector<String> Files = {
      "/file1",
      "/file2",
      "/file3"
    };

    // prepare a huge payload
    String payload_temp = String();
    for( int i=0; i<10000; i++){
      payload_temp += "0123456789";
    }

    for( int j = 0; j < 20; j++){
      for( auto & file : Files){
        const String payload = payload_temp + String(j);
        const String myFile= file;
        const String myFileTemp = file + ".tmp";

        // create and fill temp file
        File writeFile =  SPIFFS.open(myFileTemp, "w+");
        writeFile.write(reinterpret_cast<const uint8_t *>(payload.c_str()), payload.length());
        writeFile.close();
        // print file list before rename
        std::cout << FileSystem.test_returnFiles().c_str() << std::endl;
        REQUIRE( SPIFFS.exists(myFileTemp));

        // CHECK IF OLD FILE EXISTS
        if( SPIFFS.exists(myFile)){
          SPIFFS.remove(myFile);
        }

        // RENAME TMP FILE
        SPIFFS.rename(myFileTemp, myFile);

        std::cout << "iteraion: " << String(j).c_str() << std::endl;
        std::cout << FileSystem.test_returnFiles().c_str() << std::endl;
        REQUIRE_FALSE( SPIFFS.exists(myFileTemp));

      }
    }
  }
`

comino, 2017-04-24:
Please stop me, if this is going in the wrong direction. 

Turns out it also happens with only one big file (200k), but more iterations, before the issue occurs. 
It happens on Arduino Implementation and on a std::vector mocked version. 

A newly opened *tmp file cant be written anymore.