pellepl/spiffs #144: Unable to write more than 127 bytes controller goes to hard fault

sandeepmalladi, 2017-04-27:
Hello,
Thanks for proving an initiation to my project. I had interfaced stm32l073RZ controller with (8 MHz external clock frequency) an external serial nor flash mx25R (64 Mega bit). I got to find your serial flash file system after lot of efforts to integrate with my program i found it working. But i have an issue when writing the file more than 127 bytes. I have no idea what would be the reason but would like to mention in detailed way according to my debug. There are 5 sensor values that have to store on the serial by opening, creating a file if necessary, read and write in to file. After careful debug I found that Memory updating function makes it all after 127 bytes are being written.  Below mention function is making this issue. Kindly look in to the issue and also please provide me solution. May be if its regarding size issue then could you suggest me Page size, sector size and any other important factor which are required for the memory updating. I had also attached the log file to better understanding the issue. Thank you.

` ((spiffs_page_ix*)((u8_t*)objix_hdr + sizeof(spiffs_page_object_ix_header)))[data_spix] = data_page;`
`if (cur_objix_spix == 0) {
      // update object index header page
      ((spiffs_page_ix*)((u8_t*)objix_hdr + sizeof(spiffs_page_object_ix_header)))[data_spix] = data_page;
      debug("\r\nappend: %04x wrote page %04x to objix_hdr entry %02x in mem\n", fd->obj_id
          , data_page, data_spix);
      objix_hdr->size = offset+written;
    } else {
      // update object index page
      ((spiffs_page_ix*)((u8_t*)objix + sizeof(spiffs_page_object_ix)))[SPIFFS_OBJ_IX_ENTRY(fs, data_spix)] = data_page;
      debug("\r\nappend: %04x wrote page %04x to objix entry %02x in mem\n", fd->obj_id
          , data_page, SPIFFS_OBJ_IX_ENTRY(fs, data_spix));
    }`
[capture.txt](https://github.com/pellepl/spiffs/files/961064/capture.txt)

pellepl, 2017-05-03:
Hi,
it is hard to figure out what is the problem. My bet would be your spi flash driver. Since you have cache, spiffs is not dumping stuff to spi flash until you've reached your 127 byte limit, where the hardfault bites you. Try to break at your hardfault handler and do a backtrace to see exactly what is causing it.

sandeepmalladi, 2017-05-03:
Hi @pellepl ,
Thanks for you suggestion, I think as per my debug I don’t have any issue with serial flash drivers as I was able to create a small file so that able to read and write data in to it but when I use the SIFFS comes the problem. I did check with other program to read the data from serial and I am able to find the data but hard fault is happening at the point of function where spiffs_object_append in (spiffs_nucleus.c). `((spiffs_page_ix*)((u8_t*)objix_hdr + sizeof(spiffs_page_object_ix_header)))[data_spix] = data_page;`. Every time it breaks here. I have no idea how to get rid of this hard fault. When it comes to this statement immediately my compiler even get hanged and forcefully need to restart again. I would like to mention there details of memory i had provided are they ok for MX25R6432F Serial Nor Flash. Thank you. One more point is that my controller uses external oscillator so max frequency generation would be 16 Mhz. will this be any issue.

` 
 SPIFFS_CFG_PHYS_SZ            = (1024*1024*8),
 SPIFFS_CFG_PHYS_ERASE_SZ = (65536),
 SPIFFS_CFG_PHYS_ADDR =      (0),
 SPIFFS_CFG_LOG_PAGE_SZ =    (256),
 SPIFFS_CFG_LOG_BLOCK_SZ =   (65536).`

pellepl, 2017-05-03:
Hmm, ok - the speed should not be an issue. Sounds more like an alignment problem. You can try enabling
```C
#define SPIFFS_ALIGNED_OBJECT_INDEX_TABLES       1
```
in your spiffs_config.h and see if that solves it.

sandeepmalladi, 2017-05-08:
Hello @pellepl  Thank you for your suggestion and its working fine now.

pellepl, 2017-05-09:
Great!

closed 2017-05-09 by pellepl

lxrobotics, 2020-01-08:
I've run into the exact same issue using a W25Q16DV SPI flash when writing a SPIFFS integration for the **Arduino MKR MEM Shield** (https://github.com/lxrobotics/Arduino_MKRMEM).

Enabling
```C
#define SPIFFS_ALIGNED_OBJECT_INDEX_TABLES       1
```
in my `spiffs_config.h` fixed the issue.

@pellepl PS: I hereby dutifully report another usage for SPIFFS - on the Arduino MKR MEM Shield :wink:

pellepl, 2020-01-19:
👍😁