pellepl/spiffs #145: Wear leveling issue

arc3, 2017-04-27:
Hello, there.
I haven't found information on this issue, so I'm asking here.
Does wear leveling feature work correctly with constant files?
I'm testing spiffs and I've come across next situation:
presets:
phys_size = 1Mb
phys_sector_size = 64Kb
log_block_size = 64Kb
log_page_size = 1024b
1. I write some big file. Size of 500Kb.
2. I start multiple writing&removing operations for smaller file. Size of 50Kb.
3. Then I check block erases and see next:
-------------------after 499 rewriting small file 
0000 ddddd//ddddddddddddddddd///////////////////////////////////////	era_cnt: 555
0001 ddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddd	era_cnt: 0
0002 ddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddd	era_cnt: 0
0003 ddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddd	era_cnt: 0
0004 ddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddd	era_cnt: 0
0005 ddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddd	era_cnt: 0
0006 ddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddd	era_cnt: 0
0007 _______________________________________________________________	era_cnt: 561
0008 _______________________________________________________________	era_cnt: 562
0009 //////////////////////////ddddddddddddd////////////////////////	era_cnt: 558
000a /////////////////////////////ddddddddddddddddd/////////////////	era_cnt: 560
000b //dddddddddddddddddddddddddddddddddddd/ddddd///////////////////	era_cnt: 551
000c ////////////////////////////////////dddddddddddddddd///////////	era_cnt: 559
000d //////////////////////////////////////////dddddddddidddddddiidd	era_cnt: 557
000e ddddddddddddddddddddddddddddddddddddddddddddddddd______________	era_cnt: 556
000f _______________________________________________________________	era_cnt: 563

Those "era_cnt: 0" means that fully written blocks do not participate in wear leveling.
Am I right?
I've played with spiffs_gc_find_candidate() here
https://github.com/pellepl/spiffs/blob/master/src/spiffs_gc.c#L297
Removed "deleted_pages_in_block > 0" condition and GC began to include fully written blocks
into candidates list. Because of erase age part of score calculation.
After that erases became more or less the same for all blocks.
I had to increase SPIFFS_GC_MAX_RUNS also.
But, of course, total number of read&writes increased too.
So, the question is:
Is the behaviour from my example correct and expected?
p.s. "Static wear leveling works the same as dynamic wear leveling except the static blocks that do not change are periodically moved so that these low usage cells are able to be used by other data." by wikipedia :-)
p.p.s. sorry for my bad English. ))

pellepl, 2017-04-29:
Hi,

all your assumptions are correct. That's a bug you found there! Spiffs is supposed to recycle low usage cells also.

Luckily, this case will probably not happen very often - actually filling a block with data calls for having big logical page size and being able to write a full logical block of data in one go (which seldom is the case in embedded business).

What I have to figure out what the reason  was for adding `deleted_pages_in_block > 0`. Now, four years later, it seems stupid.

Thanks for finding this!

pellepl, 2017-05-02:
Well, I cannot find a single reason to why `deleted_pages_in_block > 0` was added. Since all tests pass when removing it, I'll do so.

Cheers / P

commit fec9999e (2017-05-02): fixes #145
closed 2017-05-02 by pellepl