pellepl/spiffs #146: File missing

BelMoretto, 2017-05-04:
Hi all, 
I've experienced a file unrecognizing during a "power loss test". 

The program flow was the following:

1. power up
2. perform SPIFFS_check() function 
3. check the presence of files
4. check the content of files
5. Rewrite all files (deleting the previous one)
6. return to 3 

Meanwhile intentional power_loss/restart occurs. 
I noticed that when power loss occurs during the execution of SPIFFS_checks() it can happen that a file written in the previous cycle is unrecognized. 
Is this a well known issue? 
Thanks in advance. 
M. 

MCU Kinetis KV5x Family

pellepl, 2017-05-09:
Hmm, what does unrecognized mean in this context? It disappears totally?

I guess it could happen during stage 5 if the powerloss comes precisely between the deletion of the old version and the creation of the new version. Rewrite via O_CREAT | O_TRUNC first deletes, and then creates new in a non-atomic way.

ClamentJohn, 2018-01-10:
Hey,
I'm facing a similar issue. When I **reset** the MCU the file system (occasionally) outputs an error -10004 (**ERR_DELETED**). This is not seen every time I tried out this test. Also seen when I tried the "**power loss test**".
Here's what I'm doing:
1. Mount
2. Open file (previously made)
3. Seek to the beginning of file and read                          - - - > **Fails here**
4. Seek to the beginning and write new data into flash
5. Flush
6. Go to step 3

* How to get the error number to see if opening the file was successful?
* Even though read "fails" write is successful and from then on "read" is also successful. So I'm 
  assuming opening the file was successful. This not always the case as sometimes both "read" and 
  "write" fail.