pellepl/spiffs #147: spiffs issue found during testing

budmassey, 2017-05-05:
I'm in the process of evaluating spiffs for use in an embedded design.

I'm seeing a failure when running the a test in a loop.  I've developed a test
that opens, writes, reads, verifies, closes, a deletes a series of files of different sizes.
You can see my test code below...  My test below ran 5 loops and failed on the 6th.

If I keep the sizes of my files relatively small (compared to the size amount of free space) I can loop
several times without error.  If my files size is large (as in this example 1MB) it will run a few loops error
free and then error on the large file with a Write errno = -10001 (SPIFFS_ERR_FULL) though SPIFFS_INFO
shows I have ~2.8MB of free space.  

SPIFFS_INFO  Total = 2863471, Used = 0, free = 2863471

   Creating/Testing  File Name = <File11.rtf>, Size = 524291 
   Write errno -10001
   test.c:143:test_spiffs:FAIL: Expected 0 Was -10001

void test_spiffs(void)
{
	u32 i;

	for (i=0;i<10;i++)
	{
		SPIFFS_check(&fs);
		SPIFFS_gc(&fs, 101*1024);
		xil_printf("\r\n************** Loops = %d \r\n **************\r\n",i+1);

	CreateVerifyTestFile("File1.txt", 8 * 256, Writebuf);
	CreateVerifyTestFile("File2.txt", 17, &Writebuf[0]);
	CreateVerifyTestFile("File3.bin", 111, &Writebuf[0]);
	CreateVerifyTestFile("File4.exe", 1024 * 2 + 1, &Writebuf[0]);
	CreateVerifyTestFile("File5.doc", 4 * 1024 + 7, &Writebuf[0]);
	CreateVerifyTestFile("File6.mp3", 4, &Writebuf[0]);
	CreateVerifyTestFile("File7.mp4", 44, &Writebuf[0]);
	CreateVerifyTestFile("File8.sys", 937, &Writebuf[0]);
	CreateVerifyTestFile("File11.rtf", (1024*1024+7), &Writebuf[0]);

	}

}

FYI.. I put my singleton information at the bottom of this page.

 

#ifndef SPIFFS_SINGLETON
#define SPIFFS_SINGLETON 1
#endif

#if SPIFFS_SINGLETON
// Instead of giving parameters in config struct, singleton build must
// give parameters in defines below.
#ifndef SPIFFS_CFG_PHYS_SZ
#define SPIFFS_CFG_PHYS_SZ(ignore)        (1024*1024*3)
#endif
#ifndef SPIFFS_CFG_PHYS_ERASE_SZ
#define SPIFFS_CFG_PHYS_ERASE_SZ(ignore)  (65536)
#endif
#ifndef SPIFFS_CFG_PHYS_ADDR
#define SPIFFS_CFG_PHYS_ADDR(ignore)      (0x900000)
#endif
#ifndef SPIFFS_CFG_LOG_PAGE_SZ
#define SPIFFS_CFG_LOG_PAGE_SZ(ignore)    (256)
#endif
#ifndef SPIFFS_CFG_LOG_BLOCK_SZ
#define SPIFFS_CFG_LOG_BLOCK_SZ(ignore)   (65536)
#endif
#endif

arc3, 2017-05-08:
Hello,
And what is the value of SPIFFS_GC_MAX_RUNS? Default is 5.
You, probably, need to increase it. 
GC erases SPIFFS_GC_MAX_RUNS of blocks maximum to allocate memory for file.
And your file seems to be larger than SPIFFS_GC_MAX_RUNS * SPIFFS_CFG_PHYS_ERASE_SZ.

budmassey, 2017-05-08:
I had SPIFFS_GC_MAX_RUNS set to the default 5

Can you tell me the best parameter settings.  I am just testing now but eventually  I plan on having about a 20MB file system with a maximum file size of 5MB.

Thank You

pellepl, 2017-05-09:
Hi, please see [this](https://github.com/pellepl/spiffs/wiki/Configure-spiffs/_edit#spiffs_gc_max_runs) for a more thorough explanation of SPIFFS_GC_MAX_RUNS.
Since your maximum file size is 5MB and your logical block size is 64k I would set it to (5M/64k)*2 = 160 to be safe. Given you do not have any requirements on the gc time.

budmassey, 2017-05-09:
Thanks for the info.  Didn't seem to change ...still fails.  I'm not sure what the issue is.

budmassey, 2017-05-15:
FYI.. I found a bug in my flash write driver... I've set my SPIFFS_GC_MAX_RUNS to match your equation above.  It seems to be working much better now.  The big problem was my flash driver.  I'm going to do more testing now.  

Also... I updated my spiffs_config.h to calculate the SPIFFS_GC_MAX_RUNS  based on a SPIFFS_GC_MAX_FILE_SIZE( I defined) and sector size.  Not sure if that makes sense for your code base.

Thanks for your help.

closed 2017-05-15 by budmassey