pellepl/spiffs #155: Combination with SD lib (both use the type File)

muisje, 2017-06-29:
Hey,

I wanted to add an sd card tp my project that runs on the esp8266. I already use spiffs. But there's an issue with compiling: ```
exit status 1
no match for 'operator=' (operand types are 'fs::File' and 'File')
```

Probably because both use type File for their files. Is it possible to perhaps change the name of the type?

kisielk, 2017-06-29:
It looks like you're trying to assign a value from one lib to a variable of the other? Can you show the actual code where the error is?

pellepl, 2017-06-29:
Hi, I cannot help with this, I'm afraid. This issue rather belongs to one of the esp8266 projects (the arduino one?), as the name clash seem to derive from the integration code and not spiffs itself. Spiffs is written in C only.

muisje, 2017-06-29:
Here are some pieces of code. 

I think the problem is that both use 'File' but it is different for both

```

#include <SPI.h>
#include <SD.h>

#include "FS.h"

File myFile;
bool isMoreDataAvailable();
byte getNextByte();

String subscribed_users_filename = "subscribed_users.json";

DynamicJsonBuffer jsonBuffer;

//flag for saving data
bool shouldSaveConfig = false;

//callback notifying us of the need to save config
void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

bool loadConfig() {
  File configFile = SPIFFS.open("/config.json", "r");
  if (!configFile) {
    Serial.println("Failed to open config file");
    return false;
  }

  size_t size = configFile.size();
  if (size > 1024) {
    Serial.println("Config file size is too large");
    return false;
  }

  // Allocate a buffer to store contents of the file.
  std::unique_ptr<char[]> buf(new char[size]);

  // We don't use String here because ArduinoJson library requires the input
  // buffer to be mutable. If you don't use ArduinoJson, you may as well
  // use configFile.readString instead.
  configFile.readBytes(buf.get(), size);

  StaticJsonBuffer<200> jsonBuffer;
  JsonObject& json = jsonBuffer.parseObject(buf.get());

  if (!json.success()) {
    Serial.println("Failed to parse config file");
    return false;
  }

  strcpy(telegramToken, json["telegramToken"]);

  // Real world application would store these values in some variables for
  // later use.

  Serial.print("Loaded telegramToken: ");
  Serial.println(telegramToken);
  return true;
}

bool saveConfig() {
  StaticJsonBuffer<200> jsonBuffer;
  JsonObject& json = jsonBuffer.createObject();
  json["telegramToken"] = telegramToken;

  File configFile = SPIFFS.open("/config.json", "w");
  if (!configFile) {
    Serial.println("Failed to open config file for writing");
    return false;
  }

  json.printTo(configFile);
  return true;
}

    if (text == "/bender") {
      String file_name = "bender.jpg";

      myFile = SD.open(file_name);

      if (myFile) {
        Serial.print(file_name);
        Serial.print("....");

        //Content type for PNG image/png
        String sent = bot.sendPhotoByBinary(chat_id, "image/jpeg", myFile.size(),
            isMoreDataAvailable,
            getNextByte);

        if (sent) {
          Serial.println("was successfully sent");
        } else {
          Serial.println("was not sent");
        }

        myFile.close();
      } else {
        // if the file didn't open, print an error:
        Serial.println("error opening photo");
      }
    }

bool isMoreDataAvailable(){
  return myFile.available();
}

byte getNextByte(){
  return myFile.read();
}

```

pellepl, 2017-06-29:
Have you seen this? https://github.com/esp8266/Arduino/issues/2281

closed 2017-07-18 by pellepl