pellepl/spiffs #160: Read file fails if SPIFFS_CACHE is 0

hkroe, 2017-07-28:
If the read and write cache is disabled (SPIFFS_CACHE = SPIFFS_CACHE_WR = 0), the read function returns -10016 (SPIFFS_ERR_INDEX_REF_LU). If the cache is enabled the read function works.

The example and the configuration:
[example.txt](https://github.com/pellepl/spiffs/files/1182911/example.txt)
[spiffs_config.txt](https://github.com/pellepl/spiffs/files/1182912/spiffs_config.txt)

The output from the example: 
[Output_cache_disabled.txt](https://github.com/pellepl/spiffs/files/1182933/Output_cache_disabled.txt)
[Output_cache_enabled.txt](https://github.com/pellepl/spiffs/files/1182932/Output_cache_enabled.txt)

pellepl, 2017-07-28:
Thanks for a detailed report. I'll try to reproduce it here.
Cheers / P

pellepl, 2017-07-30:
Hmm, I cannot reproduce it here. This suggests it is something weird with your spi flash hal functions. Actually, this rings a bell. I remember someone using an Arduino spiflash library (IIRC), where the lib didn't allow writing on already written bytes. This is however used in spiffs to update certain flag bits.
May I see your spi flash functions?

hkroe, 2017-07-31:
I have changed the debug output in the example.
[example_2.txt](https://github.com/pellepl/spiffs/files/1187225/example_2.txt)

As you can see in the output spiffs first writes a 0x00 to address 284 and then a 0x03. But address 284 is the last char of u8_t name[SPIFFS_OBJ_NAME_LEN];
[Rw_output_cache_disabled.txt](https://github.com/pellepl/spiffs/files/1187227/Rw_output_cache_disabled.txt)
[Rw_output_cache_enabled.txt](https://github.com/pellepl/spiffs/files/1187229/Rw_output_cache_enabled.txt)

hkroe, 2017-07-31:
I got it. First of all, I have to say that our processor is a MSP430 and our IDE is IAR. The issue has nothing to do witch the cache. I found the problem in file spiffs_nucleus.c in line 1366.
In that debug session the pointer objix_hdr points to 0xF08BC, data_spix is 0 and the header size is 29. So in this line we try to write a 16-bit value to the address 0xF08D9. But on the MSP430 this needs an alignment of 2. So in this case the MSP430 writes to address 0xF08D8.

In our case SPIFFS_ALIGNED_OBJECT_INDEX_TABLES doesn’t help.

pellepl, 2017-08-01:
SPIFFS_ALIGNED_OBJECT_INDEX_TABLES should help - but if you're using IAR I guess [this line of code](https://github.com/pellepl/spiffs/blob/master/src/spiffs_nucleus.h#L479) is not interpreted by IAR as alignment. You probably need to change this directive (and maybe others, search for attribute in the code) to something IAR can understand.

hkroe, 2017-08-01:
The quick and dirty solution was just to increase SPIFFS_OBJ_NAME_LEN by 1.
The IAR equivalent for __attribute(( aligned(x) )) is #pragma data_alignment = x. I have already changed this. But this didn’t change anything.
If I understand this correctly, __attribute(( aligned(sizeof(spiffs_page_ix)) )) just says that the struct has to start at an aligned address. But this won’t help in this line.
Have a look at [this line](https://github.com/pellepl/spiffs/blob/master/src/spiffs_nucleus.c#L1366). objix_hdr has an alignment of 2. ((u8_t *)objix_hdr) cast objix_hdr to a 8-bit pointer. So we lose every information about the alignment. Now we increase this 8-bit pointer by the header size. In the comment above this is (u8_t *)0xF08BC + 0x1D = (u8_t *)0xF08D9. Now we cast this 8-bit pointer to a 16-bit pointer (spiffs_page_ix *). At this pointer the IAR compiler doesn’t take care about the alignment.

pellepl, 2017-08-01:
> If I understand this correctly, __attribute(( aligned(sizeof(spiffs_page_ix)) )) just says that the struct has to start at an aligned address.

Actually, this also enforces not only that the struct has to start at an aligned address, but also that it must be of aligned size.

E.g., if I run this under gcc:
``` C
#include <stdio.h>

typedef struct __attribute(( aligned(4) )) {
  unsigned char a;
} test_struct; 

int main(int args, char **argv) {
  printf("%ld\n", sizeof(test_struct));
  return 0;
}
```

it will return
```
4
```