pellepl/spiffs #162: After gc, The number of pfree is getting smaller.

lianzhulin, 2017-08-18:
Hi, Boss:
Sometime, After gc, The number of pfree is getting smaller. and then need to try too many times. See below a piece of testsuite log .

And, How to setup the Garbage collecting heuristics when config with small page (64 Bytes)?
(block is 4KB)
Thanks a lot!

...
gc_check #33: run gc free_blocks:2 pfree:58 pallo:9285 pdele:22277 [31620] len:236 of 3422
[0m[36mgc_check: cleaning block 86, result 0
[0m[36m
gc_check #34: run gc free_blocks:2 pfree:57 pallo:9285 pdele:22278 [31620] len:236 of 3363
[0m[36mgc_check: cleaning block 85, result 0
[0m[36m
gc_check #35: run gc free_blocks:2 pfree:56 pallo:9285 pdele:22279 [31620] len:236 of 3304
[0m[36mgc_check: cleaning block 82, result 0
[0m[36m
gc_check #36: run gc free_blocks:2 pfree:58 pallo:9285 pdele:22277 [31620] len:236 of 3422
[0m[36mgc_check: cleaning block 81, result 0
[0m[36m
gc_check #37: run gc free_blocks:2 pfree:57 pallo:9285 pdele:22278 [31620] len:236 of 3363
[0m[36mgc_check: cleaning block 80, result 0
[0m[36m
gc_check #38: run gc free_blocks:2 pfree:56 pallo:9285 pdele:22279 [31620] len:236 of 3304
[0m[36mgc_check: cleaning block 109, result 0
[0m[36m
gc_check #39: run gc free_blocks:2 pfree:55 pallo:9285 pdele:22280 [31620] len:236 of 3245
[0m[36mgc_check: cleaning block 108, result 0
[0m[36m
gc_check #40: run gc free_blocks:2 pfree:54 pallo:9285 pdele:22281 [31620] len:236 of 3186
[0m[36mgc_check: cleaning block 107, result 0
[0m[36m
gc_check #41: run gc free_blocks:2 pfree:53 pallo:9285 pdele:22282 [31620] len:236 of 3127
[0m[36mgc_check: cleaning block 106, result 0
[0m[36m
gc_check #42: run gc free_blocks:2 pfree:52 pallo:9285 pdele:22283 [31620] len:236 of 3068
[0m[36mgc_check: cleaning block 119, result 0
[0m[36mgc_check: finished, 31547 dirty, blocks 3 free, 73 pages free, 43 tries, res 0
...

pellepl, 2017-09-03:
Sorry for belated answer.

Yes, on a scattered filesystem this may happen - though it should fix itself after a while. The reason is that when a file data page is moved from a block being gc:ed, the index page must be updated. If this index page is another block than the one being gc:ed, the old one is deleted. However the assumption is that during a gc, index pages and belonging data pages ought to end up in the same block.

There is a fix for this which actually is under development now, but it requires more or less a full rewrite of the fs.

Generally, I would not recommend 64 byte logical page size but if that works for you go ahead. How many blocks do your filesystem have?

lianzhulin, 2017-09-11:
My norflash size is 2~4MB, and the physical erase block size is 4KB, which may split it into small pages and  logic blocks.  For examples:
16B * 256,
64B * 64,
128B * 32,
256B * 16,
...
I don't know which is the better one. 
There has some small files and a few update operation in my filesystem.

pellepl, 2017-10-07:
Sorry for belated answer. I'd go with something like 128-256 byte page size, and perhaps 16-32-64k logical blocks. Reg. the GC heuristics, there probably is no global maximum. If it works fine for you now, I would not tinker with them. Otherwise, it is best to set up a usage model on your PC and use the testbench to simulate a year of usage and try to lock in what heuristics are optimal.
Unless you're doing lots of writes and removes, this might be unnecessary.