pellepl/spiffs #165: Files over 25 kB not written properly

joseasenz, 2017-08-29:
Hi guys.

Commit e92a6c32fc7d5d3e704bf1d2b7325a0383d4f369 on 14 July works fine for me.
But I have problems creating files bigger than 25 kb (roughly) with later commits.

I noticed page writing fails when it is a single byte write operation. It looks like the page has not been properly erased before writing.

This is my config:

        cfg.phys_size = 4*1024*1024;          
        cfg.phys_addr = 0;                    
        cfg.phys_erase_block = 4 * 1024;      
        cfg.log_block_size = 64 * 1024;       
        cfg.log_page_size = 256;

[spiffs_config.txt](https://github.com/pellepl/spiffs/files/1259859/spiffs_config.txt)

Regards

joseasenz, 2017-08-29:
Same problem with release 0.3.7.
Release 0.3.6 works fine

pellepl, 2017-09-03:
Hi, thanks for reporting. 

Hmm, I cannot reproduce it here. How do you create and fill your >25kB file? Could you please share a snippet?

Cheers

joseasenz, 2017-09-08:
Hello pellep, thanks for your response.

I am doing something like this:

void testboot()
{

	FSFILE* fichero = NULL;
	UINT32 iloop;
	UINT8 fbuf[256];

	fichero = FsyFileOpen("TEST.TXT", "w");

	if(fichero)
	{
		for(iloop = 0; iloop < 134; iloop++)  // 134 * 200 = 26800 bytes
		{
			FsyFileWrite(fbuf, 200, fichero);
		}
		FsyFileClose(fichero);
	}

	fichero = FsyFileOpen("TEST.TXT", "r");
        fsize = FsyFileSize(fichero);    // the function returns 26755 instead of 26800

	if(fichero)
	{
		fsize = 0;
		for(iloop = 0; iloop < 134; iloop++)
		{
			fsize += FsyFileRead(fbuf, 200, fichero);  // only 26755 bytes are read instead of 26800
		}

		FsyFileClose(fichero);
	}

If using 256 bytes instead of 200 for writing and reading it seems to work fine even with bigger files

pellepl, 2017-09-08:
Thanks, I can reproduce it. Will dig on.

pellepl, 2017-09-09:
Ok, I think I got it. I couldn't understand why the tests did not catch this, but it turned out that none of the tests were testing large files, with small chunks, without O_APPEND. Sigh...
Cheers for finding this!

commit f5e26c4e (2017-09-09): fixes #165
closed 2017-09-09 by pellepl