pellepl/spiffs #166: problem with creating file, fs not declared

beckmx, 2017-09-08:
I have been trying to save a file with spiffs but the compiler complains about this error

`error: 'fs' undeclared`

I have these headers

```
#include "esp_common.h"
#include "gpio.h"
#include "queue.h"

#include "spiffs.h"
#include <stdlib.h>
#include "spiffs_test_params.h"

#include "fcntl.h"
#include "unistd.h"
```

Anyone knows how to use spiffs or a better way to create or save data?

pellepl, 2017-09-08:
Sorry, need more info. What processor? ESP? What toolchain? Do you have more code to show? Did you do the integration yourself or not?

beckmx, 2017-09-08:
Iam using eap8266 under esp-open-sdk

beckmx, 2017-09-08:
this is my code:

`

#include "esp_common.h"
#include "hkc.h"
#include "gpio.h"
#include "queue.h"

#include "spiffs.h"
#include <stdlib.h>
#include "spiffs_test_params.h"

#include "fcntl.h"
#include "unistd.h"

xQueueHandle identifyQueue;

struct  gpio {
    int aid;
    int iid;
} gpio2;

void    led_task(void *arg) //make transfer of gpio via arg, starting as a static variable in led routine
{
    int     i,original;
    cJSON   *value;

    os_printf("led_task started\n");
    value=cJSON_CreateBool(0); //value doesn't matter
    while(1) {
        vTaskDelay(1500); //15 sec
        original=GPIO_INPUT(GPIO_Pin_2); //get original state
//      os_printf("original:%d\n",original);
        value->type=original^1;
        GPIO_OUTPUT(GPIO_Pin_2,original^1); // and toggle
        change_value(    gpio2.aid,gpio2.iid,value);
        send_events(NULL,gpio2.aid,gpio2.iid);
    }
}

static void example_read_file_posix()
{
    const int buf_size = 0xFF;
    uint8_t buf[buf_size];

    int fd = open("test.txt", O_RDONLY);
    if (fd < 0) {
        printf("Error opening file\n");
        return;
    }

    int read_bytes = read(fd, buf, buf_size);
    printf("Read %d bytes\n", read_bytes);

    buf[read_bytes] = '\0';    // zero terminate string
    printf("Data: %s\n", buf);

    close(fd);
}

static void example_read_file_spiffs()
{
    const int buf_size = 0xFF;
    uint8_t buf[buf_size];

    spiffs_file fd = SPIFFS_open(&fs, "other.txt", SPIFFS_RDONLY, 0);
    if (fd < 0) {
        printf("Error opening file\n");
        return;
[... 202 lines cut ...]
    //hkc_init("HomeACcessory");
    xTaskCreate(test_task, "test_task", 1024, NULL, 2, NULL);
    os_printf("end of user_init @ %d\n",system_get_time()/1000);
}

/***********************************************************************************
 * FunctionName : user_rf_cal_sector_set forced upon us by espressif since RTOS1.4.2
 * Description  : SDK just reversed 4 sectors, used for rf init data and paramters.
 *                We add this function to force users to set rf cal sector, since
 *                we don't know which sector is free in user's application.
 *                sector map for last several sectors : ABCCC
 *                A : rf cal    B : rf init data    C : sdk parameters
 * Parameters   : none
 * Returns      : rf cal sector
***********************************************************************************/
uint32 user_rf_cal_sector_set(void) {
    extern char flashchip;
    SpiFlashChip *flash = (SpiFlashChip*)(&flashchip + 4);
    // We know that sector size is 4096
    //uint32_t sec_num = flash->chip_size / flash->sector_size;
    uint32_t sec_num = flash->chip_size >> 12;
    return sec_num - 5;
}

`

pellepl, 2017-09-09:
I gather this issue should be under the esp-open-rtos project instead, but I think you're missing
```C
#include "esp_spiffs.h"
```

beckmx, 2017-09-09:
I added it but still:

error: 'fs' undeclared (first use in this function)

do you think I can create this variable missing?

pellepl, 2017-09-09:
As mentioned, I think it would be a lot safer asking this on the esp-open-rtos project. In their [spiffs example](https://github.com/SuperHouse/esp-open-rtos/blob/master/examples/spiffs/spiffs_example.c)
 it is NOT added, so I would not recommend it. Otherwise it should just be a matter of declaring `spiffs fd;` I guess.
I'd try getting esp-open-rtos's spiffs example up and running, and then take it from there. 
My guess is that you've missed including a library in the make config or something.

IngYessine, 2018-12-10:
Hello , I try do use SPIFFS on my project , I'm using IAR WORKBENCH , but when I compiled my project , some errors for headers files are not found ! exemple for this headers files "fcntl.h" and "unistd.h"
I use Windows OS !

Someone can help me ?!