pellepl/spiffs #169: Lots of reads, even with plenty of cache

markrages, 2017-09-14:
I am trying to determine good parameters for an 8 MB filesystem, with 256-byte page erases.  I plan to store a few dozen large files on it.

My filesystem is at the other end of a wireless link, so reads are expensive.  But I have plenty of RAM, so I allocated cache equal to the memory size.

My test is to mount a blank filesystem and read the directory.

Results
----------
filesystem size = 8388608

| log_block_size | log_page_size | reads | read_bytes | writes | written_bytes |
| -------------- | ------------- | ----- | ---------- | ------ | ------------- |
| 16384 | 256 | 1536 | 263168 | 0 | 0 |
| 16384 | 512 | 1536 | 525312 | 0 | 0 |
| 16384 | 1024 | 1536 | 1049600 | 0 | 0 |
| 32768 | 256 | 768 | 131584 | 0 | 0 |
| 32768 | 512 | 768 | 262656 | 0 | 0 |
| 32768 | 1024 | 768 | 524800 | 0 | 0 |
| 65536 | 256 | 640 | 131328 | 0 | 0 |
| 65536 | 512 | 384 | 131328 | 0 | 0 |
| 65536 | 1024 | 384 | 262400 | 0 | 0 |
| 131072 | 256 | 576 | 131200 | 0 | 0 |
| 131072 | 512 | 192 | 65664 | 0 | 0 |
| 131072 | 1024 | 192 | 131200 | 0 | 0 |
| 262144 | 256 | 544 | 131136 | 0 | 0 |
| 262144 | 512 | 160 | 65600 | 0 | 0 |
| 262144 | 1024 | 96 | 65600 | 0 | 0 |

Why so many reads?  The filesystem is blank.

robert-b, 2017-09-18:
Yes, i found the problem.

Id | Time | EraseCount | BytesRead | BytesWritten | ReadCallCount | WriteCallCount
---- | ---- | ---- | ---- | ---- | ---- | ---- 
1 | 3159 | 0 | 1573376 | 47 | 6146  |2
2 | 3162 | 0 | 1573646 | 47 | 6152  |2
3 | 3156 | 0 | 1570332 | 47 | 6144  |2
4 | 3163 | 0 | 1574231 | 47 | 6165  |2
5 | 3156 | 0 | 1570194 | 47 | 6155  |2
6 | 3166 | 0 | 1575072 | 47 | 6179  |2
7 | 3168 | 0 | 1575567 | 47 | 6190  |2
8 | 3168 | 0 | 1575882 | 47 | 6197  |2
9 | 3169 | 0 | 1576152 | 47 | 6203  |2
10 | 3170 | 0 | 1576422 | 47 | 6209  |2
11 | 3171 | 0 | 1576692 | 47 | 6215  |2
12 | 3171 | 0 | 1576917 | 47 | 6220  |2
13 | 3172 | 0 | 1577232 | 47 | 6227  |2
14 | 3172 | 0 | 1577502 | 47 | 6233  |2
15 | 3173 | 0 | 1577772 | 47 | 6239  |2
16 | 3176 | 0 | 1578042 | 47 | 6245  |2

i use a fresh formatted 8mb nor flash.
i just open/create a file and write a 4 byte bom.

is this normal ?
or a bug iteration over the index pages 
?

pellepl, 2017-10-07:
Sorry for belated answer.

@markrages Am I understanding this correctly, you're actually streaming the data i/o over wireless link??? 

Spiffs reads a lot, yes. This is due to the design goal of using as little ram as possible - compared to other fs:s, there are no index tables or file trees built in memory.

Mounting spiffs means reading all LU pages per block to collect statistics. Number of LU pages depends on your config.
Creating a file means reading all LU pages per block to find a unique id and avoid name clashes.

The cache is very primitive and can hold at most up to 32 pages.  This will be remedied in 0.4.0, but still a fresh mount will need to scan all LU pages to collect statistics. Also, on a fresh mount the cache is empty, so it must be filled before having any effect.