pellepl/spiffs #172: spiffs_page_delete relies on 0 -> 1 transitions being ignored

rojer, 2017-09-22:
I am working with a flash controller that does not like it when application tried to set bits from 0 to 1.
naturally, it's not possible, and most controllers just ignore it, but not this one - it raises an error for such writes, and I found that SPIFFS is a bit sloppy in this regard.

It seems that `spiffs_page_delete` tries to set the flags blindly, assuming that 0 -> 1 transitions will be ignored. Well, that is not always the case and SPIFFS should not assume this behavior of the underlying controller.

rojer, 2017-09-22:
https://github.com/pellepl/spiffs/pull/173 is my quick whack at it. i'm not at all sure this is all that needs to be done, but tests are now passing without the "flags exemption".

pellepl, 2017-10-07:
Sorry for belated reply.
Transitions 0->1 being ignored is one of the main presumptions when writing spiffs. What flash controller is it? I have never stumbled upon one not allowing this.

I am deliberately sloppy here as, in my view, this is an optimization.
Looking at your fix I am actually a bit surprised, I think you covered it all. I thought there would be more cases.
What I would like to have is a config define around the readout as it will affect performance somewhat. At least to my experience, 99% of all flashes do not have a problem with ignoring 0->1.

rojer, 2017-10-07:
controller is the one built into TI CC3220, but i think it's generic to TI (Tiva chips have the same).
it's described in chapter 21 of the [TRM](http://www.ti.com/lit/ug/swru465/swru465.pdf).
in particular, bit 10 of the `FCRIS` register is:
```
INVDRIS -- Invalid Data Raw Interrupt Status
...
1h = An interrupt is pending because a bit that was previously
programmed as a 0 is now being requested to be programmed as a
1.
```
and indeed controller refuses the write of any of the data loaded into the data registers tries to flip bits from 0 to 1.

yes, it saves a read. makes sense to make it a config option, sure.

rojer, 2017-10-07:
what would you call the option? i'll amend #173.

pellepl, 2017-10-07:
Ummm.... That is a very good question. _CONFIG_READB4WRITE? I'm open to
suggestions. 😉

pellepl, 2017-10-07:
Off topic, but the CC32 flash, that is the internal flash rom you're referring to? Not anything external to the uC itself?

rojer, 2017-10-07:
`SPIFFS_CONFIG_NO_BLIND_WRITES` ? with the default of 0, meaning blind writes are allowed and 0 -> 1 transitions are ignored.

yes, i'm using part of internal flash for the filesystem.

pellepl, 2017-10-07:
SPIFFS_CONFIG_NO_BLIND_WRITES ?

Splendid! Much better.

Ok, internal flashes seem to have more weird constraints in general. The
EFR only allows two writes to same address before one must erase IIRC.

rojer, 2017-10-08:
@pellepl ok, please take a look at #173

closed 2017-10-15 by pellepl