pellepl/spiffs #176: Reading and writing at the same time

spe-ciellt, 2017-10-04:
Hello!

I am currently building an application that runs under an RTOS (actually scmRTOS). I have defined the macros SPIFFS_LOCK and SPIFFS_UNLOCK to use OS mutexes to handle reading and writing from several tasks.

I have two task (actually many), one is reading a file from SPIFFS and another that is writing to SPIFFS. When I run both reading and writing at the same time it seems that SPIFFS fails. All of a sudden I get error -10008, the file is closed. I haven't closed it though. Sometimes it is reading the file and sometimes it is writing the file. 

Is there a way to debug this or check the file system to what is actually happening? I have been staring at bits for days and have not come any closer to any solution or idea of what is happening. Any tips or hints would be grateful, I am quite desperate.

Thanks and best regards,
/spe

pellepl, 2017-10-07:
Hi, it should work (tm). The file descriptors should definitely not get sporadically closed.
You might want to enable the debug prints in spiffs_config.h, specifically SPIFFS_API_DBG, SPIFFS_DBG and SPIFFS_CACHE_DBG.

spe-ciellt, 2017-10-09:
Thanks for input, I will try these debug flags.