pellepl/spiffs #177: Error when writing large files with chunks smaller than 256 bytes

vad32m, 2017-10-07:
I've just integrated SPIFFS int my project, target MCU is stm32f407 and memory IC is W25Q32. My application requires working with large (up to 448 kB) files. I was testd SPIFFS with 10kB files and everything seemed to be hunky-dory. But then I switched to testing my app with almost 32kB files and now I am facing following issue: if I write this file with chunks less than 256 bytes, some data loss occur. I cant disclose all the code due to some limitations but here is self-explanatory code snippet:
```
 SPIFFS_gc(&fs, 32768);
  // create a file, delete previous if it already exists, and open it for reading and writing
  fd = SPIFFS_open(&fs, "/update/firmware", SPIFFS_CREAT | SPIFFS_TRUNC | SPIFFS_WRONLY, 0);
  if (fd < 0) {
    printf("errno %i\n", SPIFFS_errno(&fs));
    return 0;
  }
	
  int32_t firmware_size = 0;
  const int32_t one_burst_write_count = 256;
	while (firmware_size < received){
		  int32_t read_count = received - firmware_size;
		  if (read_count > one_burst_write_count){
				  read_count = one_burst_write_count;
			}
                        if (SPIFFS_write(&fs, fd, &firmware_buf[firmware_size],read_count) < 0){
				 printf("write errno %d\r\n", SPIFFS_errno(&fs));
			}
      firmware_size += read_count;
  }
  
  if (SPIFFS_close(&fs, fd) < 0) {
    printf("errno %i\n", SPIFFS_errno(&fs));
    return 0;
  }
  
  fd = SPIFFS_open(&fs, "/update/firmware", SPIFFS_RDONLY, 0);
  if (fd < 0) {
    printf("errno %i\n", SPIFFS_errno(&fs));
    return 0;
  }
  
  printf("firmware file \r\n");
  firmware_size = 0;
  //readout file and check it
  while (!SPIFFS_eof(&fs,fd)){
      SPIFFS_read(&fs, fd, (u8_t *)buf, 1);
      if (buf[0] != firmware_buf[firmware_size]){
          printf("%x error %d != %d\r\n", firmware_size, buf[0], firmware_buf[firmware_size]);
      }
      firmware_size++;
  }
  
  printf("filesize %d\r\n",firmware_size);
  
  if (SPIFFS_close(&fs, fd) < 0) {
    printf("errno %i\n", SPIFFS_errno(&fs));
    return 0;
  }
```
It works OK when `one_burst_write_count ` is 256 or greater, but if I set it to number, less than 256, SPIFFs wont properly write (or read) file. When it will perform file checks errors will appear, data in file beginning from some address will not correspond to data from the `firmware_buf`, which is actually source for file. Errors with text like `6774 error 135 != 143` will appear (see printf statement) and file size calculated with `firmware_size` variable will be little lower. For example file size, when `one_burst_write_count = 132` is 0x6777 and file size is 31855 bytes instead required 31900. If I dump that file binary into console it seems that everything is OK, except missing 45 (31900 - 31855) bytes, strting at 0x6777 file offset. If I modify chunk size to 255 errors will still persist, but they will start from offset 0x67F2 and file size will be 31735 bytes. Setting chunk size to 256 fixes the problem. Can anyone approve that this is bug, or maybe I made mistake somewhere?
BTW `firmware_buf` is initialized with binary content, you may found in archive.
[firmware.zip](https://github.com/pellepl/spiffs/files/1366314/firmware.zip)

pellepl, 2017-10-07:
Hi, thanks for a detailed report. I'll look it through more thoroughly when
I've got some more time at hand.
However, this really should work and have been tested. Similar problems
before have in 90 perc of all cases turned out to be a bug in the spiflash
driver or in the integration.
In meantime, I'd recommend you to really scrutinize your spi flash driver
code and make yourself absolute certain that it does what it should. Also,
if you're running in a threaded context, make sure the mutex locking works.

Cheers /P

pellepl, 2017-10-07:
Oh, and what's your config?

vad32m, 2017-10-09:
@pellepl It is quite possible that there are errors in my implementation. I attached archive with my files, so you can see SPIFFS config as well, as my driver implementation. And here is one more snippet, that performs initialization
```
spiflash_t winbond_flash_handler;
static spiffs fs;

#define LOG_PAGE_SIZE       256

static u8_t spiffs_work_buf[LOG_PAGE_SIZE*2];
static u8_t spiffs_fds[32*4];
static u8_t spiffs_cache_buf[(LOG_PAGE_SIZE+32)*4];

static s32_t my_spiffs_read(u32_t addr, u32_t size, u8_t *dst) {
    return SPIFLASH_read(&winbond_flash_handler, addr, size, dst);
 }

static s32_t my_spiffs_write(u32_t addr, u32_t size, u8_t *src) {
    return SPIFLASH_write(&winbond_flash_handler,addr, size, src);
}

static s32_t my_spiffs_erase(u32_t addr, u32_t size) {
    return SPIFLASH_erase(&winbond_flash_handler, addr, size);
} 
  
 
void my_spiffs_mount() {
    spiffs_config cfg;
    
    cfg.hal_read_f = my_spiffs_read;
    cfg.hal_write_f = my_spiffs_write;
    cfg.hal_erase_f = my_spiffs_erase;
    
    int res = SPIFFS_mount(&fs,
      &cfg,
      spiffs_work_buf,
      spiffs_fds,
      sizeof(spiffs_fds),
      spiffs_cache_buf,
      sizeof(spiffs_cache_buf),
      0);
     

 printf("mount res: %i\n", res);
}
```
[SpiffsBugReportFiles.zip](https://github.com/pellepl/spiffs/files/1367817/SpiffsBugReportFiles.zip)
Will appreciate any help or advice.

pellepl, 2017-10-10:
Ah, you're using my spiflash driver. Then I can't really blame it on you can I? :) 
Will see things through.

ch1648, 2018-10-22:
Any update on this issue?   I am facing the same issue.

Here is my test case:

FS Page size is 256

#define TEST_DATA_LENGTH           25000
#define PATTERN_BUF_LENGTH      10

1)     Format the File System   <<<<<<<This is need to make it happen every time.
2)     fd = FILE_Open(TEST_FILE_NAME, FILE_CREAT | FILE_RDWR, 0);

3) Write to file a test pattern 10 bytes a time till 25000

    for (k=0; k<TEST_DATA_LENGTH; k+=PATTERN_BUF_LENGTH)
    {
      ret = FILE_Write(fd, test_pattern, sizeof(char)*PATTERN_BUF_LENGTH);
      if (ret != sizeof(test_pattern))
          PRINTF("Write file failed, result %d, expected %d, location %d\r\n", ret, sizeof(test_pattern), k);
      }
    }
    
    FILE_Close(fd);

4) Read back the file, and the pattern comparison failed at offset 24357

I looked into the log.   The error happens in spiffs_object_append call

      **when** cur_objix_spix != prev_objix_spix   and per_objix_spix == 0
   
      when object index header page is out of room, and about to use new objix index page (?)

You can see from the log the file reading jumping from page C2 to C9.

Data copying (moving) in file writing after page C2, and somehow error happens.

**Is there any document on how to handle the case that object index header page is out of room?**
Let me know if you needs more information.

Thank you,
Chris

Here is a section of log for writing

.....
SPIFFS_write 1 10<LF>
CACHE_WR_WRITE: storing to cache page 1 for fd 1:8001, offs 24470:220 len 10<LF>
SPIFFS_write 1 10<LF>
CACHE_WR_WRITE: storing to cache page 1 for fd 1:8001, offs 24480:230 len 10<LF>
SPIFFS_write 1 10<LF>
CACHE_WR_WRITE: storing to cache page 1 for fd 1:8001, offs 24490:240 len 10<LF>
SPIFFS_write 1 10<LF>
CACHE_WR_DUMP: dumping cache page 1 for fd 1:8001, boundary viol, offs:24250 size:250<LF>   <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
append: 250 bytes @ offs 24250 of size 24250<LF>
append: 8001 load objixhdr page 00c3:0000<LF>
CACHE_ALLO: allocated cache page 2 for pix 00c3<LF>
CACHE_FREE: free cache page 3 pix 00c0<LF>
CACHE_ALLO: allocated cache page 3 for pix 00c2<LF>
append: 8001 store to existing data page, 00c2:0060 offset:154, len 97, written 0<LF>
append: 8001 wrote page 00c2 to objix_hdr entry 0060 in mem<LF>
append: 8001 store objix 00c3:0000, written 97<LF>
CACHE_FREE: free cache page 2 pix 00c3<LF>
       CALLBACK  HUP obj_id:8001 spix:0000 npix:00c4 nsz:24347<LF>
       callback: setting fd 1:8001(fdoffs:24500 offs:24250) objix_hdr_pix to 00c4, size:24347<LF>
       callback: setting fd 1:8001 span:0000 objix_pix to 00c4<LF>
append: 8001 store new objix_hdr, 00c4:0000, written 97<LF>
       CALLBACK  NEW obj_id:8001 spix:0001 npix:00c5 nsz:0<LF>
append: 8001 create objix page, 00c5:0001, written 97<LF>
append: 8001 store new data page, 00c6:0061 offset:0, len 153, written 97<LF>
append: 8001 wrote page 00c6 to objix entry 0000 in mem<LF>
[... 236 lines cut ...]
CACHE_ALLO: allocated cache page 0 for pix 1d00<LF>
CACHE_FREE: free cache page 3 pix 1b01<LF>
CACHE_ALLO: allocated cache page 3 for pix 1d01<LF>
CACHE_FREE: free cache page 1 pix 1c00<LF>
CACHE_ALLO: allocated cache page 1 for pix 1e00<LF>
CACHE_FREE: free cache page 2 pix 1c01<LF>
CACHE_ALLO: allocated cache page 2 for pix 1e01<LF>
CACHE_FREE: free cache page 0 pix 1d00<LF>
CACHE_ALLO: allocated cache page 0 for pix 1f00<LF>
CACHE_FREE: free cache page 3 pix 1d01<LF>
CACHE_ALLO: allocated cache page 3 for pix 1f01<LF>
CACHE_FREE: free cache page 1 pix 1e00<LF>
CACHE_ALLO: allocated cache page 1 for pix 0000<LF>
CACHE_FREE: free cache page 2 pix 1e01<LF>
CACHE_ALLO: allocated cache page 2 for pix 0001<LF>
CACHE_FREE: free cache page 0 pix 1f00<LF>
CACHE_ALLO: allocated cache page 0 for pix 00ca<LF>
read: load objix page 00ca:0001 for data spix:0061<LF>
read: offset:24347 rd:251 data spix:0061 is data_pix:00c9 addr:6830c905<LF>
CACHE_FREE: free cache page 3 pix 1f01<LF>
CACHE_ALLO: allocated cache page 3 for pix 00c9<LF>
read: offset:24598 rd:249 data spix:0062 is data_pix:00cb addr:6830cb05<LF>
CACHE_FREE: free cache page 1 pix 0000<LF>
CACHE_ALLO: allocated cache page 1 for pix 00cb<LF>
Read back error, result = 0, expected 7<CR><LF>