pellepl/spiffs #181: 0.37 -- Crash on write

feistyz, 2017-10-23:
[spiffs_config.h.txt](https://github.com/pellepl/spiffs/files/1405852/spiffs_config.h.txt)

[crash.txt](https://github.com/pellepl/spiffs/files/1405850/crash.txt)

I am trying to integrate spiff into an Arduino Zero based board with the [MX66L51235F](http://www.macronix.com/Lists/Datasheet/Attachments/6233/MX66L51235F,%203V,%20512Mb,%20v1.1.pdf) Flash.
I am able to compile everything and run it. The file system mounts ok and I can verify that the flash is getting written to properly.

However, when I try to create a "Hello World" file per integration example from the wiki the library crashes in spiffs_nucleus.c at line 1308 which is:
((spiffs_page_ix*)((u8_t *)objix_hdr + sizeof(spiffs_page_object_ix_header)))[data_spix] = data_page;

I think that's a red herring as there is nothing wrong with that line and all the variables are within the right range of the buffer it is trying to access. I also see that the flash gets "Hello World" written to it. I think something happens earlier (probably some overrun) that then causes this to crash.

It is possible I have something misconfigured though it's not obvious. I have turned on all the DBG prints in the library but nothing obvious is jumping out. The output of the DBG print is attached which starts from a clean Flash with everything erased and then does a SPIFFS_mount and tries to write the file. Test code is below.

Any insight would be great.

```
void spiffs_mount() {
  spiffs_config cfg;
  cfg.phys_size = 1*1024*1024; // use all spi flash
  cfg.phys_addr = 0; // start spiffs at start of spi flash
  cfg.phys_erase_block = 65536; // according to datasheet
  cfg.log_block_size = 65536; // let us not complicate things
  cfg.log_page_size = LOG_PAGE_SIZE; // as we said
  
  cfg.hal_read_f = spiffsRead;
  cfg.hal_write_f = spiffsWrite;
  cfg.hal_erase_f = spiffsErase;

  int res = SPIFFS_mount(&fs,
    &cfg,
    spiffs_work_buf,
    spiffs_fds,
    sizeof(spiffs_fds),
    spiffs_cache_buf,
    sizeof(spiffs_cache_buf),
    0);

  Serial.print("Mount: "); Serial.println(res);
}

static void test_spiffs() {

  Serial.println("test_spiffs");

  Serial.println("TEST -- Read/Write Files");
  char buf[12];

  Serial.println("TEST -- Open File To Write");
  // create a file, delete previous if it already exists, and open it for reading and writing
  spiffs_file fd = SPIFFS_open(&fs, "Test1", SPIFFS_CREAT | SPIFFS_TRUNC | SPIFFS_RDWR, 0);
  if (fd < 0) {
    Serial.print("TEST -- Open Error"); Serial.println(SPIFFS_errno(&fs));
    return;
  }

  Serial.println("TEST -- Write To File");
  // write to it
  if (SPIFFS_write(&fs, fd, (u8_t *)"Hello world", 12) < 0) {
    Serial.print("TEST -- Write Error"); Serial.println(SPIFFS_errno(&fs));
    return;
  }

  Serial.println("TEST -- Close File");
  // close it
  if (SPIFFS_close(&fs, fd) < 0) {
    Serial.print("TEST -- Close Error"); Serial.println(SPIFFS_errno(&fs));
    return;
  }

  Serial.println("TEST -- Open File to Read");
  // open it
  fd = SPIFFS_open(&fs, "my_file", SPIFFS_RDWR, 0);
  if (fd < 0) {
    Serial.print("TEST -- Open Error"); Serial.println(SPIFFS_errno(&fs));
    return;
  }

  Serial.println("TEST -- Read File");
  // read it
  if (SPIFFS_read(&fs, fd, (u8_t *)buf, 12) < 0) {
    Serial.print("TEST -- Read Error"); Serial.println(SPIFFS_errno(&fs));
    return;
  }

  Serial.println("TEST -- Close File");
  // close it
  if (SPIFFS_close(&fs, fd) < 0) {
    Serial.print("TEST -- Close Error"); Serial.println(SPIFFS_errno(&fs));
    return;
  }
  
  Serial.println(buf);
}

void doSPIFF() {
  spiffs_mount();
  test_spiffs();
}

```

pellepl, 2017-10-23:
try enabling SPIFFS_ALIGNED_OBJECT_INDEX_TABLES

feistyz, 2017-10-23:
That didn't help. I also tried disabling caching which also had no effect.

feistyz, 2017-10-23:
Anything else I can try to track this down?

pellepl, 2017-10-23:
Hmm. Did you try putting debug prints around you HAL layer, more specifically before and after the spiffsWrite functionality? Perhaps you have a stack smash or something. Logwise, all looks ok so I am a bit bewildered I must admit.

feistyz, 2017-10-23:
Do you mean debug prints from the call back read/write functions? If so yes. See attached file with spiffs and my debugs. I show the first 20byts of what it's trying to read/write. But it all looks kosher :-(

[crash_with_HAL_trace.txt](https://github.com/pellepl/spiffs/files/1408238/crash_with_HAL_trace.txt)

feistyz, 2017-10-23:
I have also tried:
1. Open and close the file without writing anything. This works fine and has no crash.
2. Write zero-length buffer to the file. This also works with no crash
3. Write various different length to the file: 256, 512, 1024, 64k -- All of these cause a crash after it writes the first 256 bytes.

feistyz, 2017-10-23:
I have also disabled the actual write to flash when the write to the test file is issued (just to take the flash write code out of the loop) but it still crashes in the same way :/

pellepl, 2017-10-23:
Ok, that is really weird. Is this in a threaded context? How much stack is
there? What kind of crash is it, e.g. bus error or invalid op?

feistyz, 2017-10-23:
I agree it is quite odd. 

I can't really hook anything up to it to tell what type of crash it is (I am limited to printf).
If you are asking if this is running in a multi-threaded setup then no. 
On the Arduino Zero, the stack uses whatever memory is left over after everything is loaded and allocated. At the time it crashes there are 27K of free space.

pellepl, 2017-10-23:
Hmm...
What if you do this
```  C
  Serial.println("TEST -- Open File To Write");
  // create a file, delete previous if it already exists, and open it for reading and writing
  //////////spiffs_file fd = SPIFFS_open(&fs, "Test1", SPIFFS_CREAT | SPIFFS_TRUNC | SPIFFS_RDWR, 0);
  spiffs_file fd = SPIFFS_open(&fs, "Test1", SPIFFS_CREAT | SPIFFS_TRUNC | SPIFFS_RDWR | SPIFFS_O_DIRECT, 0);
  if (fd < 0) {
    Serial.print("TEST -- Open Error"); Serial.println(SPIFFS_errno(&fs));
    return;
  }

  Serial.println("TEST -- Write To File");
  // write to it
  //////////if (SPIFFS_write(&fs, fd, (u8_t *)"Hello world", 12) < 0) {
  const char *data = "Hello world";
  if (SPIFFS_write(&fs, fd, (u8_t *)data, 1) < 0) {
    Serial.print("TEST -- Write Error"); Serial.println(SPIFFS_errno(&fs));
    return;
  }
```

feistyz, 2017-10-23:
Did you mean for the size to be "1"?

pellepl, 2017-10-23:
Yep

feistyz, 2017-10-23:
Ok. It crashes in the exact same spot.

pellepl, 2017-10-23:
Ok. If you delete the file directly after you created it? No writing. Does
that crash?

feistyz, 2017-10-23:
Let me try that. Also did you want me to keep SPIFFS_ALIGNED_OBJECT_INDEX_TABLES set on or off? The previous test was with it off.

pellepl, 2017-10-23:
Looks like it doesn't matter. But keep it on to rule out things for now.
Btw, what compiler are you using?

pellepl, 2017-10-23:
Stupid question, but the flash is erased right?

feistyz, 2017-10-23:
Valid question. Flash is erased. Do you want me to erase it on every test run we do?
Compiler: arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.8.3 20140228 (release) [ARM/embedded-4_8-branch revision 208322]

Deleting the file works fine. I did a SPIFFS_open and then a SPIFFS_fremove.
I also did SPIFFS_open, SPIFFS_close and SPIFFS_remove(filename) which also worked.

pellepl, 2017-10-23:
Well, compiler is old but should definitely work. Flash should be erased yes. Hmmm... you could try enabling SPIFFS_MAGIC and SPIFFS_MAGIC_LEN and do a proper format but I doubt it will change anything. Running out of ideas I'm afraid. Perhaps you could try disable compiler optimisation. 
It shouldn't be stack smashing, but you could try disabling prints also to be on safe side (though I guess you already done that)

feistyz, 2017-10-23:
you are correct formatting didn't help.
But I changed compiler options from -Os (code size) to -O0 (compile time) and now it is not crashing. That seems to suggest the compiler is mangling/removing code that it thinks is unused?

feistyz, 2017-10-23:
Crap sorry... I jumped the gun. I had the write commented out. It's still crashing :/

pellepl, 2017-10-23:
Good in a way that gcc is not to blame though. Sorry, I can't come up with anything else right now. Need to sleep a wee bit.

feistyz, 2017-10-23:
All good. Thanks for the help. I will try to hook up a debugger to it and see what I can come up with.

feistyz, 2017-10-24:
Well... it's for sure nutty. I have gotten a debugger attached and everything works without any crashing (of course). To get the debugger attached I had to switch to Windows (from a Mac) so now I am thinking this is some sort of compiler/core library issue since the Windows core binaries are slightly different. I will keep digging to see if I can figure it out as it would be nice not to be chained to my Windows box :-)

Thank you for the help!

Related to this I have a performance question. The Flash I am using is 64MBytes, with 64k blocks and 256 byte pages. I need to store a few small files as well as a few large ones (>30MB). When I write one of the large files 4K at a time I get ~180KB/sec. If I just write to the Flash raw (without spiff) I get about ~280KB/sec. If I then delete the large files the delete operation takes forever (>20min) full flash erase takes ~90sec. When I watch the delete in action it does a ton of page readings a few writes and even less flash level erasing. All of the above is done with a freshly formatted flash (in other words empty)

Really two questions:
1. Is there a better config to get faster writes?
2. What can I do to speed up the delete of a large file?

feistyz, 2017-10-25:
To put some number on it.
Delete 1MB file takes 150sec (2.5min)
Delete 10MB files takes 1490sec (25min)

pellepl, 2017-10-26:
Sorry for belated reply. Did you find the reason of the crash? Hopefully your back in the comfy Linux zone again 😉

Deleting and erasing are two different things. Delete marks all pages of a file as deleted. In order to do so, all pages must be found. Hence all reading and writing.

I would recommend having two filesystems, one with large pages for your big files, and one with small pages for the others. 
Actually, if possible, I would try to have the big files on raw flash.
Perhaps you can store the metadata of such big raw flash blobs in a spiffs file and get around it that way?
 Are these firmware images?

feistyz, 2017-10-27:
No worries... Thankful for the help.

I haven't figured it out yet as it will involve systematically downgrading various libraries until it starts working, which is zero fun, so I have been procrastinating :/

Yup, the delete vs. erase makes perfect sense. Maybe in a future version, there would be a way to optimize some of it. But for now, I have pretty much implemented what you suggested. Two file systems one using spiffs to keep track with and the other doing direct writes to the Flash. It all seems to work (for the moment).

The large files are streaming sensor data which I am shoving onto the flash as fast as it will let me.