pellepl/spiffs #186: SPIFFS_open returns SPIFFS_ERR_DELETED

bugadani, 2017-11-18:
My basic use-case of SPIFFS basically consists of creating a file at powerup, writing data to it and the powering off. The files are created with the current timestamp in the name, with an algorithm that makes the name is unique if it isn't. Files are created with SPIFFS_CREAT | SPIFFS_RDWR flags set.

After littering the FS with a whole bunch of create/remove pairs, my currently empty (meaning no valid files) FS won't create any more files, SPIFFS_open just returns SPIFFS_ERR_DELETED.

Digging into the issue with the debugger, the return code is set in spiffs_nucleus.c Line 1202, `  SPIFFS_VALIDATE_OBJIX(oix_hdr.p_hdr, fd->obj_id, 0);`

I could just erase the whole flash when I clean up my files but that takes a long-long time. Also, according to SPIFFS_info, my flash is basically empty, with 32M total, 55k used. How could I make SPIFFS understand that there are plenty of free pages to chose from?

Thanks

feistyz, 2017-11-29:
You could try running SPIFFS_check if your mount fails and see if it fixes anything.

bugadani, 2017-12-05:
In my case the issue is that SPIFFS_open creates the file header in an address using spiffs_object_create (concrete example, 0x1001500), then validates it in spiffs_object_open_by_page, but this function reads the heaeder back from an incorrect address (0x1500, in this concrete case).

bugadani, 2017-12-05:
All right, my issue is that the type spiffs_page_ix is 16 bit but my flash has like 130k pages so it is basically a configuration error.

ClamentJohn, 2018-01-11:
Hey. 
In my case, SPIFFS returns SPIFFS_ERR_DELETED when I try reading a file. It does not return an error when I try opening the file.
Heres what I do:
1. Open file
2. Read file
3. Write file
4 Flush

If I go to write directly instead of reading then it is successful. Only read fails! Any clue?

bugadani, 2018-02-15:
You can also try to check your variable sizes and if that is not your issue, please open a new issue.

closed 2018-02-15 by bugadani

samcaulfield-ubotica, 2020-04-28:
> All right, my issue is that the type spiffs_page_ix is 16 bit but my flash has like 130k pages so it is basically a configuration error.

Thank you for mentioning this! I have been using SPIFFS on a flash chip with 128 MiB storage capacity and couldn't figure out why I wasn't able to create very large files. Turns out I forgot to change the type of spiffs_page_ix to one that can represent enough indices for the entire flash.