pellepl/spiffs #190: Integrate SPIFFS example - No error but no data read either

ClamentJohn, 2017-12-14:
No errors received.
I've followed the Integrate SPIFFS [tutorial](https://github.com/pellepl/spiffs/wiki/Integrate-spiffs) to use it with my NOR flash (AT45DB041), I use an STM32F103.  But when I run the example it doesn't read anything or write anything.
I've tried printing every data that reaches my driver and they are not strings so I receive a bunch of '?'.
I wrote the driver for this and I assume this works properly (I tried to write, read and chip erase) 

AT45DB - 4Mbit data flash, one bock is 2KB
Configuration
`cfg.phys_size = 4*1024*1024; // use all spi flash
  cfg.phys_addr = 0; // start spiffs at start of spi flash
  cfg.phys_erase_block = 16*1024; // according to datasheet
  cfg.log_block_size = 16*1024; // let us not complicate things
  cfg.log_page_size = LOG_PAGE_SIZE; // as we said`

The output I receive
`errno after write  = 0
errno after close = 0
errno after read  = 0
-->  <--`
What could I be missing?

ClamentJohn, 2017-12-15:
I hadn't set `SPIFFS_USE_MAGIC` now that I've set it I'm unable to mount it. I followed the config from the wiki and have config'ed it according to my NOR flash. And for formating I've mount, unmount, format, mount. But still, no file system is seen in the flash.

pellepl, 2017-12-15:
Hi John,

We're in the middle of a heavily delayed deadline at work and I have no
time right now I'm afraid.
However, everything you mention indicates that your spiflash functions are
not working. My best bet would be to verify that you actually can
read/write/erase the spiflash.

Cheers/ P

<https://github.com/pellepl/spiffs/issues/190#issuecomment-351990606>, or mute
the thread

ClamentJohn, 2018-01-09:
Hey, thank you for the help. It was a hiccup in my write function. Closing the Issue.

closed 2018-01-09 by ClamentJohn