pellepl/spiffs #200: SPIFFS_ERR_IS_FREE when repeatedly opening & writing same file

lampmaker, 2018-01-21:
I'm running SPIFFS on an Atmel 8-bit AVR device. 

basic stuff seems to work well (open, read, write, delete, etc)  but when running a longer test I get errors.
Basically, I'm repeatedly opening the same file (flags  SPIFFS_CREAT | SPIFFS_TRUNC | SPIFFS_RDWR )  and writing the same data. Not too much, about 200-300 bytes every time.
After a couple (4-5 )of times, speed drops and I get SPIFFS_ERR_IS_FREE errors.
running SPIFFS_check returns 0 and does not fix issues.  Only a format fixes things.  any ideas?

512 Kb flash memory
#define SPIFFS_SINGLETON 1
#define SPIFFS_CFG_PHYS_SZ		524288
#define SPIFFS_CFG_PHYS_ERASE_SZ  4096
#define SPIFFS_CFG_PHYS_ADDR      0
#define SPIFFS_CFG_LOG_PAGE_SZ    256
#define SPIFFS_CFG_LOG_BLOCK_SZ   4096
#define SPIFFS_TEMPORAL_FD_CACHE              0
#define SPIFFS_IX_MAP                        0

lampmaker, 2018-01-21:
ah, bug in SPI erase ...  working fine now

closed 2018-01-21 by lampmaker

A-R-S-D, 2021-12-07:
Hi, There!

> I'm running SPIFFS on an Atmel 8-bit AVR device.

which AVR MCU did you used? i'm trying to use Arduino Nano and the compiled binary size overflows flash entire space which is 32KB.