pellepl/spiffs #204: SPIFFS_ERR_DELETED,SPIFFS_ERR_END_OF_OBJECT

MostafaASG, 2018-01-30:
Hi
I'm working on an embedded system. My system collects data from many parts like GPS, ADXL, etc and every second system creates an event with size of 80 bytes. Every time an event is created, I write it to a winbond W25Q128BV SPI Flash chip. I open the file once at system startup and read/write event in the file. Since I create every second an event and write to the file, it has dynamic size. I wanted to create a circular write process for my file, therefore I declare a counter variable to know how many event has been written in SPIF. I use this counter to seek the location in file. At the startup, system works fine but after creating some events I get SPIFFS_ERR_END_OF_OBJECT for seeking. Sometimes I seek without any error and I write data in spi flash with no error too but after write command when I want to flush, it returns SPIFFS_ERR_DELETED error. Some times only SPIFFS_ERR_DELETED error for flush command shows up and. I get these errors when I seek with SPIFFS_SEEK_SET mode. When I seek at the end of the file I barely get errors. I don't what's going on! My circular write function is added in blow.

`
void write_event_report_circular (event_report_typeDef *ev_re)
{
  printf("writing event/report\n");
  if(ev_re->event_report_type == EVENT_TYPE){
    printf(" Code:%02x\n",ev_re->event_report.event.event_code);
    if(ev_re->event_report.event.event_code == 0xAA) //offline
    {
      gps_temp.max_gps_speed_offline=0;
      ev_re->not_sent = 0;
      ev_re->not_sent_src = 0;
    }
    else
    {
      gps_temp.max_gps_speed=0;
      ev_re->not_sent = 1;
      ev_re->not_sent_src = 1;
    }
  }
  else
  {
    ev_re->not_sent = 1;
    ev_re->not_sent_src = 1;
    printf(" Code:%02x\n",ev_re->event_report.report.report_code);
  }
  ev_re->id = event_report_id++;

  if(SPIFFS_lseek(&fs , fd_event_report , event_report_wr_index * sizeof(event_report_typeDef) , SPIFFS_SEEK_SET) < 0)
  {
    printf("Seek event errno %i\n", SPIFFS_errno(&fs));
    printf("evre wr index %i\n", event_report_wr_index);
    printf("evre location %i\n", event_report_wr_index * sizeof(event_report_typeDef));
  }
  if (SPIFFS_write(&fs , fd_event_report , ev_re , sizeof(event_report_typeDef) ) < 0)
    printf("Write enent errno %i\n", SPIFFS_errno(&fs));
  if (SPIFFS_fflush(&fs, fd_event_report) < 0)
    printf("Write enent errno %i\n", SPIFFS_errno(&fs));

  event_report_wr_index++;
  if(event_report_wr_index >= MAX_NUM_EVENTS_REPORTS)//MAX_NUM_EVENTS_REPORTS = 100,000
  {
    event_report_wr_index = 0;
    event_report_turn_over ++;
  }
  if(event_report_counter_circular != MAX_NUM_EVENTS_REPORTS)
    event_report_counter_circular++;
  
  spiffs_stat stat; 
  SPIFFS_stat(&fs,"event_report.bin",&stat);
  event_report_counter = stat.size / sizeof(event_report_typeDef);
  printf("\nfilesize:%d , count:%d , size_evre:%d\n",stat.size,event_report_counter,sizeof(event_report_typeDef));
`
Do you have any idea why I get these errors?

pjsg, 2018-01-30:
Can you convert this code into a testcase and run it as part of the self tests for SPIFFS? If you get the same error, then that makes debugging much easier. If you don't get the same error, then it could be your flash driver or some other issue in your code.

bugadani, 2018-02-15:
Please check if your data sizes are large enough for your memory chips. I had a similarly bizarre issue because I did not check that SPIFFS used 16 bit variables for my 32mbit memory.

MostafaASG, 2018-02-21:
@pjsg, I did what you said and again same errors received.
@bugadani, What do you mean? Would you please explain about "SPIFFS used 16 bit variables for my 32mbit memory"?

I wrote a test code. It is added in below.
`
////////////////////////// Test part /////////////////////////////////////////
  test_index ++;
  printf("!!!!Test index is set to:!!!!%i\n", test_index);
  int32_t ress = SPIFFS_lseek(&fs , fd_SPITEST , (test_index-1) * sizeof(test_index) , SPIFFS_SEEK_SET);
  if(ress < 0)
  {
    printf("Seek errno %i\n", SPIFFS_errno(&fs));
    printf("$$$SPI Test write part$$$\n");
    printf("Test Index:%i\n", test_index);
    printf("test location %i\n", (test_index-1) * sizeof(test_index));
  }
  else
    printf("***********Seek*********  %i\n", ress);
  
  printf("going to write test_index in the file\n");
  if (SPIFFS_write(&fs , fd_SPITEST , &test_index , sizeof(test_index) ) < 0)
    printf("Write enent errno %i\n", SPIFFS_errno(&fs));
  else
    printf("data:%i is written in the test file\n", test_index);
  
  //////////////////////
  printf("going to seek to read test_index from the file\n");
  ress = SPIFFS_lseek(&fs , fd_SPITEST , (test_index-1) * sizeof(test_index) , SPIFFS_SEEK_SET);
  if(ress < 0)
  {
    printf("Seek errno %i\n", SPIFFS_errno(&fs));
    printf("###SPI Test read part###\n");
    printf("Test Index:%i\n", test_index);
    printf("test location %i\n", (test_index-1) * sizeof(test_index));
  }
  else
    printf("***********Seek*********  %i\n", ress);
  
  if(SPIFFS_read(&fs,fd_SPITEST,&read_data,sizeof(test_index))<0)
  {
    printf("reading errno %i\n", SPIFFS_errno(&fs));
  }
  else
    printf("data:%i is read from the test file\n", read_data);
  //////////////////////////////////////////////////////////////////////////////
` 
Every time I run this code, after some iterations, for example 'test_index == 1000', when I seek to address '999*4(byte)' to write the variable in the SPIF, I get 'SPIFFS_ERR_END_OF_OBJECT' error. After this error when I increment the 'test_index' in the next iteration and try to write it in the SPIF, the number that is written in the SPIF is the last successful number that is written in the SPIF without any error in seeking the address. In this example, after test_index==1000, instead of incremented 'test_index", every number that is written in the SPIF is 999. That's weird.

Do you have any idea what's happening to my SPIF?

bugadani, 2018-02-21:
Size of various indexing variables are set here: https://github.com/pellepl/spiffs/blob/master/src/default/spiffs_config.h#L360

I would guess that your test code indeed writes the values correctly and it is the read part that fails and returns the last value it can access. The value 1000 doesn't seem special, is this the actual value where your code fails?

MostafaASG, 2018-02-21:
My logical page size is set to 256 bytes. I tried 128 bytes but there is no difference. I get the same error. 1000 is not special number that occurs every time. It was just an example. The real numbers in my test are 444 or 4570 or any random number.

MostafaASG, 2018-02-22:
@bugadani This is my terminal log

!!!!Test index is set to:!!!!391
***********Seek*********  1560
***********Seek*********  1560
going to write test_index in file
***********Seek*********  1560
***********Seek*********  1560
data:391 is read from the test file

!!!!Test index is set to:!!!!392
***********Seek*********  1564
***********Seek*********  1564
going to write test_index in the file
data:392 is written in the test file
going to seek to read test_index from file
***********Seek*********  1564
***********Seek*********  1564
data:392 is read from the test file

!!!!Test index is set to:!!!!393
***********Seek*********  1568
***********Seek*********  1568
going to write test_index in the file
data:393 is written in the test file
going to seek to read test_index from file
***********Seek*********  1568
***********Seek*********  1568
data:393 is read from the test file

!!!!Test index is set to:!!!!394
***********Seek*********  1572
***********Seek*********  1572
going to write test_index in the file
data:394 is written in the test file
going to seek to read test_index from the file
***********Seek*********  1572
***********Seek*********  1572
data:394 is read from the test file

!!!!Test index is set to:!!!!395
***********Seek*********  1576
***********Seek*********  1576
going to write test_index in the file
data:395 is written in the test file
going to seek to read test_index from the file
***********Seek*********  1576
***********Seek*********  1576
data:394 is read from the test file

!!!!Test index is set to:!!!!396
Seek errno -10003
$$$SPI Test write part$$$
Test Index:396
test location 1580
going to write test_index in the file
data:396 is written in the test file
going to seek to read test_index from the file
***********Seek*********  1580
***********Seek*********  1580
data:394 is read from the test file

!!!!Test index is set to:!!!!397
Seek errno -10003
$$$SPI Test write part$$$
Test Index:397
test location 1584
going to write test_index in the file
data:397 is written in the test file
going to seek to read test_index from the file
***********Seek*********  1584
***********Seek*********  1584
data:394 is read from test file

!!!!Test index is set to:!!!!398
Seek errno -10003
$$$SPI Test write part$$$
Test Index:398
test location 1588
going to write test_index in the file
data:398 is written in the test file
going to seek to read test_index from the file
***********Seek*********  1588
***********Seek*********  1588
data:394 is read from the test file

MostafaASG, 2018-04-08:
Hi, my write index made problem and I handle that now. Thank you all.

closed 2018-04-08 by MostafaASG