pellepl/spiffs #207: SPIFFS_write error 10010 when creating spiffs image

mrfaptastic, 2018-03-22:
Hello,

When attempting to use the mkspiffs @ https://github.com/igrr/mkspiffs to generate an spiffs image, the tool fails to build the spiffs image. @igrr 

Example:
```
user@pc:~$ ./mkspiffs --size 1004 --page 256 --block 8192 -d 5 --create ./test spiffsfile
Debug output enabled
/testin.txt
file size: 1088
SPIFFS_write error(-10010): unknown
```

Looking at spiffs.h, -10010 relates to:
#define SPIFFS_ERR_BAD_DESCRIPTOR       -10010

I don't understand why this error is occurring as there are certainly no issues with the device, and spiffs code only appears to error here.

igrr, 2018-03-22:
You're setting the image size to 1004 bytes? That's even smaller than the block size..

mrfaptastic, 2018-03-23:
I don't have a clue what I'm doing to be honest. Generally I use the Arduino IDE and the 'ESP8266 Sketch Data Upload' add-on. When I use this, it prints the size, page and block values to the console.... I just assumed I could copy these values as input argument for this tool. Clearly not :-)

`[SPIFFS] data   : C:\XXXXXX\ESP8266Project\data
[SPIFFS] size   : 1004
[SPIFFS] page   : 256
[SPIFFS] block  : 8192
/advanced.html
/config.html.gz
/login.html
/message.html.gz
/success.html

[SPIFFS] upload : C:\......\build2cf67cbbce69f5af8e76208b05c18086.spiffs/ESP8266Project.spiffs.bin
[SPIFFS] reset  : nodemcu
[SPIFFS] port   : COM3
[SPIFFS] speed  : 921600
[SPIFFS] address: 0x300000

Uploading 1028096 bytes from <removed> to flash at 0x00300000

................................................................................ [  7% ]

etc.`

closed 2021-04-07 by mrfaptastic