pellepl/spiffs #210: SPIFFS_ERR_DELETED

MostafaASG, 2018-04-11:
Hi
Sometimes when I try to read or write to a file, I get 'SPIFFS_ERR_DELETED' error. When I get this error, I can no longer work with my file and this error shows up repeatedly. After a while, SPIFFS fails and it makes me to format the SPI Flash.
I open 4 files with dynamic size at the same time, but read and write a lot to just one file.

Any idea what's going on with my SPIFFS?

jgeissberger, 2020-05-28:
Hi Mostafa,

Were you able to identify the problem and have you found a fix for it?

I have started to encounter the same error on a project I am working on.

MostafaASG, 2020-05-29:
Hi
Using this library is a big mistake. It's completely waste of time. Listen my friend. If you want to save sth. just select the hard way and write your own SPI Flash library. It's not hard. Just read your SPI Flash datasheet carefully. Working with SPI Flash with low level library gives you freedom. Using SPI Flash with this file system library is a deadend.

jago85, 2020-05-30:
This software is a masterpiece. I'm using it for years without trouble. How can some compare low level flash access to a filesystem with wear leveling and caching? I don't say that there may not be any bugs. But, trying to rebuild all this functionality on your own would be a deadend.

I'd like to thank the author of this software for publishing all the hard work here for free.

jcwren, 2020-05-30:
Gotta go with @jago85 here. Had some initial teething pains as we learned our way around it and adapted it to our flash device (problems in our driver), but it's been rock solid. Sounds like a case of sour grapes for @MostafaASG.

pellepl, 2020-05-30:
@jago85  @jcwren Thanks guys. That was a heartwarmer ♥️

jgeissberger, 2020-05-30:
I just wanted to follow up on that.

In the end, I noticed that the error was on my side, as I had implemented the block-erase function wrong (messed up the sector size), so instead of erasing the 64kB I had planned (and configured), I used the function to erase 4kB sectors, so only after a while (and seemingly at random), I would encounter this Error, as the FS would find those half-deleted blocks and wouldn't know what to do with it.

After fixing the error, I ran a mass-rewrite operation without any problems, so all is good now!

pellepl, 2020-05-30:
👍

valkuc, 2020-05-30:
Using this library in esp8266 project as a file-based database for more than 3 years. All ok, all good. So if someone don’t like elephants, he simply don’t know how to cook them.

pellepl, 2020-05-31:
@jago85

samcaulfield-ubotica, 2020-06-04:
Just to add some hopefully useful information to anyone searching for this error message: similar to @jgeissberger I encountered the error SPIFFS_ERR_DELETED when using logical page sizes greater than the flash physical page size due to an error in my implementation of the hardware abstraction layer. Specifically, I had an incorrect implementation of the SPI write function for the flash chip I am using with SPIFFS.

While I'm here, thank you for writing this library @pellepl!