pellepl/spiffs #215: Random loss/corruption of data when writing (errno -10004)

boreycutts, 2018-07-02:
So I am not very experienced in working with NOR Flash memory or file systems so bear with me. I have somehow managed to create a 1MB spiffs partition on my NOR Flash chip with the following configuration:
```
cfg.phys_size = 1 * 1024*1024;              // Partition size       : 1MB
cfg.phys_addr = 0x50800000;                 // Partition location   : 0x50800000
cfg.phys_erase_block = 256 * 1024;          // Erase Block Size     : 256kB
cfg.log_block_size = 256 * 1024;            // Block Size           : 256kB
cfg.log_page_size = LOG_PAGE_SIZE;          // Page Size            : 256B

typedef u16_t spiffs_block_ix;
typedef u16_t spiffs_page_ix;
typedef u16_t spiffs_obj_id;
typedef u16_t spiffs_span_ix;
```
With this I am able to create small files and read them back however my implementation requires the ability to write ~50kB files. Writing these files gives me no errors however when I read the contents of memory there are missing file headers and missing/corrupt data. When this occurs the read function will typically give the error -10004. This occurs very randomly so it has been difficult to figure out exactly what's happening.  

I'm almost certain that it's not a driver error because I was able to write a 50kB string to the flash and read it back. However I do think that it has something to do with my configuration because when I change `spiffs_page_ix` to 32 bit, writing works perfectly but when I try to open the file I get the error -10072 which doesn't seem to correlate to any of the defined error codes in spiffs.h. If more info is needed just let me know.

boreycutts, 2018-07-05:
Upon further debugging I have noticed that the files are intact immediately after writing. However when i execute `SPIFFS_read()`, when I check the contents of the memory through the debugger, in random locations there will be missing/corrupted data. There is typically a series of `0x00`,  `0x55` (`0b01010101`) or `0xaa` (`0b10101010`) bytes where the original data was. Once again I tested this only using driver operations and no corruption occurred so I don't think it's my drivers.

boreycutts, 2018-07-09:
Ok my bad I realized it was a problem with the cache. I disabled caching during reads and the file system works perfectly. Apparently I'm not flushing the cache correctly so if anyone has any advice on how to do so let me know.

pellepl, 2018-07-14:
This seems more like a memory corruption to me. Probably something, not being spiffs, is writing in your cache buffers. Oxaa and 0x55 feels like specific patterns to me, could this give you a clue of what it could be?
You can try moving the cache buffers in memory or assigning the double amount of cache memory just to test.

fvpalha, 2018-08-01:
@boreycutts , What is the solution? How flush the cache properly? This is important for the developer community.

boreycutts, 2018-08-01:
@fvpalha So I'm using spiffs on top of the MQX RTOS which has its own function for flushing the cache `DCACHE_FLUSH()`. I call that function after every NOR flash read and so far there hasn't been any more data corruption.