pellepl/spiffs #221: write array to flash

devy13, 2018-08-23:
hello
i am using CYC8CKIT - 050 cypress kit and W25Q128FV flash chip and implemented SPIFFS
now i want to write and read BMP file array to/from flash chip using spiffs_write and tried to do that but not able to read actual image

every write and read to/from flash give different errors including -10003,-10004,-10005,-10014,-10010
i am attaching my code for review 

what will be the reason for that?
any help?
[TFT_SSD1963.cywrk.Archive05.zip](https://github.com/pellepl/spiffs/files/2314411/TFT_SSD1963.cywrk.Archive05.zip)
al image

devy13, 2018-08-27:
solved the problem
it was in typedef

i have working sd card routine for display image from sd card
anyone have idea how can i store/write bmp file in flash and display bmp on tft from flash?

devy13, 2018-08-31:
anyone here?
i am not able to read write and append data without SPIFFS_TRUNC
what might be the reason and how to solve it?

tonila, 2019-06-12:
It's been a while, but I had similar issue in Simplelink device, that I could only write data with SPIFFS_TRUNC flag. 
Solution was to increasing stackSize.