pellepl/spiffs #224: Check if file is already open?

lampmaker, 2018-09-28:
Hi,

I've been going trough the docs but couldn't find an answer to this:

what happens if I call spiffs_open() twice with the same path+filename ? Do I get two different filehandles to the same file?
Is there any way to prevent or detect this? I have a situation where a file might or might not be open. If it is not open, it should be opened, if it is open, the existing filehandle should be used.

If course, I can write some code to check the names of open files, but if there is an easy way to do this with SPIFFS, that would have my preference..

Thanks!

pellepl, 2018-09-30:
You get two different filehandles to the same file, yes. There's nothing implemented to detect or prevent this - in fact, this is the way it is meant to work.

lampmaker, 2018-09-30:
Thanks!
I'll see if I find a way to implement something. Perhaps add a flag to spiffs_open that tells it to return an existing filehandle if the file is already open,  and do the regular stuff it it's not open yet.  I'll see.   Need to find some time first :-)

closed 2018-09-30 by lampmaker