pellepl/spiffs #228: SPIFFS_open() fails after format with SPIFFS_ERR_FULL

eyalasko, 2018-11-27:
Hi, 

I'm trying to implement SPIFFS but can't pass beyond mounting.

my test code does the following:
mount > unmount > format > re-mount
Then I try to SPIFFS_open(&fs, "file1", SPIFFS_CREAT | SPIFFS_TRUNC | SPIFFS_RDWR, 0);
and get SPIFFS_ERR_NOT_FOUND (-10002) error code.

My SPI flash (Adesto [AT45DB641E](https://www.adestotech.com/wp-content/uploads/DS-45DB641E-027.pdf)) physical block/page size is 256KB/256Bytes (1024 pages/block)
I'm Using 4 Blocks (1MB) starting at address 0.

Modifications to spiffs_config.h from default config are:
- SPIFFS_BUFFER_HELP  1
- SPIFFS_GC_STATS  0
- SPIFFS_USE_MAGIC/SPIFFS_USE_MAGIC_LENGTH   1
- SPIFFS_SINGLETON  1
- SPIFFS_CFG_PHYS_SZ    (4 * 1024 * 256)
- SPIFFS_CFG_PHYS_ERASE_SZ    (1024*256)
- SPIFFS_CFG_LOG_PAGE_SZ    (256)
- SPIFFS_CFG_LOG_BLOCK_SZ    (1024*256)

The test code is:
'      
       int spiffs_test(void)
       {
	int ret;

	static spiffs fs;
	static u8_t spiffs_work_buff[SPIFFS_CFG_LOG_PAGE_SZ() * 2];
	static u8_t spiffs_fds[/*sizeof(spiffs_fd)*/ 48*4];	// profiling using SPIFFS_buffer_bytes_for_filedescs()
	static u8_t spiffs_cache_buf[(SPIFFS_CFG_LOG_PAGE_SZ()+20)*4 + 20]; // profiling using SPIFFS_buffer_bytes_for_cache()
	
	// mount
	spiffs_config cfg;

    cfg.hal_read_f = spiflash_read_simple;
    cfg.hal_write_f = spiflash_write_simple;
    cfg.hal_erase_f = spiflash_erase_simple;
    
    ret = SPIFFS_mount(&fs, &cfg, spiffs_work_buff, spiffs_fds, sizeof(spiffs_fds), spiffs_cache_buf, sizeof(spiffs_cache_buf), NULL);
    SPIFFS_unmount(&fs);
    ret = SPIFFS_format(&fs);
    ret = SPIFFS_mount(&fs, &cfg, spiffs_work_buff, spiffs_fds, sizeof(spiffs_fds), spiffs_cache_buf, sizeof(spiffs_cache_buf), 0);		
	
    const char str1[] = "Hello spiffs";
	
    spiffs_file fd = SPIFFS_open(&fs, "file1", SPIFFS_CREAT | SPIFFS_TRUNC | SPIFFS_RDWR, 0);  // << fd = -10002
    ret = SPIFFS_write(&fs, fd, (u8_t *)str1, strlen(str1));
    ret = SPIFFS_close(&fs, fd); 
'

Any advise is highly appreciated

michaelkollmann, 2019-01-22:
Sorry to bother you but I'm facing the same issue.
Have you come across a solution?

Also, no response from the project owner since you posted back in november. Is this project dead?

eyalasko, 2019-01-22:
> Sorry to bother you but I'm facing the same issue.
> Have you come across a solution?
> 
> Also, no response from the project owner since you posted back in november. Is this project dead?

Hi, 

Unfortunately, I'm still facing a same/similar issues with opening a file.

The project's creator posted the following notice on October 2017 : 

> 0.4.0 is under construction. This is a full rewrite and will change the underlying structure. Hence, it will not be compatible with earlier versions of the filesystem. The API is the same, with minor modifications. Some config flags will be removed (as they are mandatory in 0.4.0) and some features might fall away until 0.4.1. If you have any worries or questions, it can be discussed in issue [#179](https://github.com/pellepl/spiffs/issues/179)

I have recently posted similar inquiry on stackoverflow ,  still without response  :( ([https://stackoverflow.com/questions/54240958/spiffs-open-fails-after-format](url)

Please update if you find a solution/workaround.

Cheers

michaelkollmann, 2019-01-22:
> Please update if you find a solution/workaround.

Will do. Thanks for your quick reply!

michaelkollmann, 2019-01-23:
Found a solution!

In my case the solution was pretty simple. My SPI driver was only supporting data with a length defined by a uint8_t (255B). My page size was defined as 256B tho.

All I had to do was to keep the page size at 256B in the config but in the bottom layer (HAL) to separate each read and write into two separate calls, offset by 128B.

eyalasko, 2019-01-23:
OK, I see it now.

Did as you suggested (split >255 bytes message to 2 consecutive messages). However, I still can't get the file to open. I suspect that there's something wrong with the spiffs definitions.
Can you share your spiffs_config.h?
What type of spi flash are you using?

Thanks

ShaharHD, 2020-03-10:
Hi,
I don't know if it's related or not ... 

I'm trying to use the `nrfx_qspi` library as the backend to the `spiffs` - but I keep getting `NRF_ERROR_INVALID_ADDR` due to `nrfx_is_word_aligned(p_rx_buffer)` - which comes from the `spiffs` crazy offset calculation for the work buffer

- I'm working with the DevKit on-board flash - which was verfied to work fine  with the FatFS demo from nordic...

iMazstick, 2020-10-12:
Hi @eyalasko,

I know this is an old thread, I am also using a nRF52 device and I was wondering if you ever managed to solve your issue. I applied @michaelkollmann 's fix and I still can't get past mounting. I'd appreciate any help or advice you may have. I'm using a W25Q80DV flash device.

Thanks

eyalasko, 2020-10-13:
Hi @iMazstick,

It's been a while but if I recall correctly @michaelkollmann advice solved the problem.
I tried to look at some old code but unfortunately we have never implemented this back into the main application so I can't get trace of the change (try to look at the SPI drivers though)

iMazstick, 2020-10-13:
Hi @eyalasko,

Thanks for taking time to respond so quick, I really appreciate it. I have done what @michaelkollmann mentioned but I still can't get past mounting the file system. I will persevere with it for a while and see what happens.

Thanks

hoai265, 2021-08-01:
Hi guys, any update for this issue?
I also got this issue although i used the static buffer for testing
`
uint32_t spiffs_api_mount() {
    /* Do the mount */
    cfg.hal_read_f = spiffs_api_read;
    cfg.hal_write_f = spiffs_api_write;
    cfg.hal_erase_f = spiffs_api_erase;

    int res = SPIFFS_mount(&fs, &cfg, spiffs_work_buf, spiffs_fds, sizeof(spiffs_fds), spiffs_cache_buf, sizeof(spiffs_cache_buf), 0);
    TRACE_INFO("SPIFFS_mount err=%d", res);
    if (res != SPIFFS_OK) {
        if (res == SPIFFS_ERR_NOT_A_FS) {
            res = SPIFFS_format(&fs);
            TRACE_INFO("SPIFFS_format err=%d", res);
            if (res) {
                return res;
            }
            return SPIFFS_mount(&fs, &cfg, spiffs_work_buf, spiffs_fds, sizeof(spiffs_fds), spiffs_cache_buf, sizeof(spiffs_cache_buf), 0);
        } else {
            SPIFFS_unmount(&fs);
            return res;
        }
    } else {
        SPIFFS_unmount(&fs);
    }

    return SPIFFS_OK;
}

uint32_t spiffs_api_create_file(const char* filePath) {
    s32_t ret = SPIFFS_creat(&fs, filePath, 0);
    return ret < 0;
}
`
`
#define SPIFFS_BUFFER_SIZE (512*1024)
uint8_t SPIFFS_BUFFER[SPIFFS_BUFFER_SIZE];

s32_t spiffs_api_read(u32_t addr, u32_t size, u8_t* dst) {
    TRACE_INFO("spiffs_api_read addr=%d size=%d", addr, size);
    if (addr + size > SPIFFS_BUFFER_SIZE) {
        return 1;
    }
    memcpy(dst, SPIFFS_BUFFER + addr, size);
    return 0;
}

s32_t spiffs_api_write(u32_t addr, u32_t size, u8_t* src) {
    TRACE_INFO("spiffs_api_write addr=%d size=%d", addr, size);
    if (addr + size > SPIFFS_BUFFER_SIZE) {
        return 1;
    }
    memcpy(SPIFFS_BUFFER + addr, src, size);
    return 0;
}

s32_t spiffs_api_erase(u32_t addr, u32_t size) {
    TRACE_INFO("spiffs_api_erase addr=%d size=%d", addr, size);
    if (addr + size > SPIFFS_BUFFER_SIZE) {
        return 1;
    }
    memset(SPIFFS_BUFFER + addr, 0, size);
    return 0;
}
`

hoai265, 2021-08-01:
Hi guys, any update for this issue?
I also got this issue in SPIFFS_creat although i used the static buffer for testing

```
uint32_t spiffs_api_mount() {
    cfg.hal_read_f = spiffs_api_read;
    cfg.hal_write_f = spiffs_api_write;
    cfg.hal_erase_f = spiffs_api_erase;
    int res = SPIFFS_mount(&fs, &cfg, spiffs_work_buf, spiffs_fds, sizeof(spiffs_fds), spiffs_cache_buf, sizeof(spiffs_cache_buf), 0);
    TRACE_INFO("SPIFFS_mount err=%d", res);
    if (res != SPIFFS_OK) {
        if (res == SPIFFS_ERR_NOT_A_FS) {
            res = SPIFFS_format(&fs);
            TRACE_INFO("SPIFFS_format err=%d", res);
            if (res) {
                return res;
            }
            return SPIFFS_mount(&fs, &cfg, spiffs_work_buf, spiffs_fds, sizeof(spiffs_fds), spiffs_cache_buf, sizeof(spiffs_cache_buf), 0);
        } else {
            SPIFFS_unmount(&fs);
            return res;
        }
    } else {
        SPIFFS_unmount(&fs);
    }
    return SPIFFS_OK;
 }

uint32_t spiffs_api_create_file(const char* filePath) {
    s32_t ret = SPIFFS_creat(&fs, filePath, 0);
    return ret < 0;
}
```

driver 
```
#define SPIFFS_BUFFER_SIZE (512*1024)
uint8_t SPIFFS_BUFFER[SPIFFS_BUFFER_SIZE];
s32_t spiffs_api_read(u32_t addr, u32_t size, u8_t* dst) {
    TRACE_INFO("spiffs_api_read addr=%d size=%d", addr, size);
    if (addr + size > SPIFFS_BUFFER_SIZE) {
        return 1;
    }
    memcpy(dst, SPIFFS_BUFFER + addr, size);
    return 0;
}

s32_t spiffs_api_write(u32_t addr, u32_t size, u8_t* src) {
    TRACE_INFO("spiffs_api_write addr=%d size=%d", addr, size);
    if (addr + size > SPIFFS_BUFFER_SIZE) {
        return 1;
    }
    memcpy(SPIFFS_BUFFER + addr, src, size);
    return 0;
}

s32_t spiffs_api_erase(u32_t addr, u32_t size) {
    TRACE_INFO("spiffs_api_erase addr=%d size=%d", addr, size);
    if (addr + size > SPIFFS_BUFFER_SIZE) {
        return 1;
    }
    memset(SPIFFS_BUFFER + addr, 0, size);
    return 0;
}

```

ShaharHD, 2021-08-01:
@hoai265  could this be related to #264 which I encountered a couple of years ago?

hoai265, 2021-08-01:
@ShaharHD Thanks for quick reply bro.
I just test with static buffer as above so the word alignment with not my problem i thought.
But i will give [ littefs](https://github.com/littlefs-project/littlefs) a try.

ShaharHD, 2021-08-01:
> @ShaharHD Thanks for quick reply bro.
> I just test with static buffer as above so the word alignment is not my problem i thought.
> But i will give [ littefs](https://github.com/littlefs-project/littlefs) a try.

The main reason was the response I got on #264 which simply said the project is not maintained. littlefs on the other hand is being actively maintained and development by ARM (soon to be Nvidia?)

hoai265, 2021-08-02:
Finally found the problem in my code. 
The issue is in my erase code
```
s32_t spiffs_api_erase(u32_t addr, u32_t size) {
    TRACE_INFO("spiffs_api_erase addr=%d size=%d", addr, size);
    if (addr + size > SPIFFS_BUFFER_SIZE) {
        return 1;
    }
    memset(SPIFFS_BUFFER + addr, 0, size);
    return 0;
}

```
It is my mistake to erase value to 0x00:
`memset(SPIFFS_BUFFER + addr, 0, size);`
I change it to
`memset(SPIFFS_BUFFER + addr, 0xFF, size);`
and the problem solved.