pellepl/spiffs #233: SPIFFS_ERR_IS_FREE - how to find the cause of failure ?

ranshalit, 2019-02-27:
Hello,

I hope someone can help.
I am struglling with integrating spiffs with stm32.
I manage to open, write, close, (all return value are ok) , but then on continue a second open and read(), I get SPIFFS_ERR_IS_FREE , as if previous write was not successful (?)

Is there a way to find our what's the problem ?

Thanks

BertVanoverschelde, 2019-04-10:
I had a similar issue like you, my problem was located in the SPI driver, not in the spiffs code. If you want to debug this, you can try setting the spiffs work buffer and spiffs cache buffer volatile and globally declared, then add watchpoint to these two buffers. check that the first few bytes correspond to the correct object index and the correct span index, and that the flags byte is correct.

kisielk, 2019-04-10:
If you're using DMA, ensure your buffers are word aligned in memory. If you're using a processor like one of the STM32F7 which has a data cache make sure to flush the cache before starting a DMA TX and then invalidate the cache after a DMA RX. Or use the DTCM memory.

siworks, 2021-06-07:
Hi, 
I have a similar problem. Also on stm32f407, I manage to mount the FS. If I first write to the file and then read from it, it works ok, but if I just try to read from the file after mounting I get the SPIFFS_ERR_INDEX_REF_FREE error.
I am also using SPI. @BertVanoverschelde , can you please tell some more details, what was wrong with your SPI driver and how you fixed it?

Thanks