pellepl/spiffs #241: User defined hook for long running operations

josesimoes, 2019-08-02:
First: a **BIG** thank you for this awesome project! 👏 

I'm using it for some large (and not so large) flash chips and I've run into some issues when performing long running operations like format. 
When flash operations take a considerable time (as expected) my watchdog is kicking in and resetting the CPU.

Was thinking that long running operations could include in the loops that responsible for those long running operations a simple call to a user defined function.
This would be a simple define that's empty as default.
If needed the user can implement it's function with whatever content works for it's use case. In my particular situation it would be a call to the API responsible to feed the watchdog.

I can submit a PR with this, just wanted to throw this idea for comments.

pellepl, 2019-08-02:
Just a thought: If it is only formatting and GC you're having in mind, the user hal function for erasing blocks could include that callback.

josesimoes, 2019-08-03:
Interesting... I'll give it a try.

josesimoes, 2019-08-29:
This works perfectly! Thank you so much for the suggestion. 👏 💯

closed 2019-08-29 by josesimoes