pellepl/spiffs #264: word Alignment

ShaharHD, 2020-03-11:
Been working on incorporating the library into nRF52840 (Nordic) test project

- Flash device: `MX25R6435F` (on the DevKit)
- Connected via QSPI

Been getting a lot of error on word alignment, but the most problematic one is inside `spiffs_page_move` the code requests via `hal_write_f` to write a page of **1 byte**

As the common QSPI flash are word aligned (32bit) AFAIK, is this 1 byte write by design?
Is there a flag somewhere to control and make sure everything is word aligned in read/write operations?

ShaharHD, 2020-04-06:
Is this project in active development?
will a demo project for Nordic DevKit would help?

pellepl, 2020-04-06:
No pretty dormant. One byte writes are per design see wiki.

ShaharHD, 2020-04-07:
Maybe worthing mentioning it on the main README.md - alignment is a big thing on embedded devices which uses QSPI flash ...

closed 2021-08-01 by ShaharHD

jago85, 2021-08-02:
Hi, I'm curious what's the actual problem here. QSPI flashs don't have word/dword alignment. At least I've never seen one. You can write a single byte to any given address.
How do you come to this conclusion? Even your device MX25R6435F doesn't mention anything like this in the datasheet.

ShaharHD, 2021-08-02:
The nRF528xx chip has a QSPI interface which is 4 byte aligned (pretty standard).
SPI Flash is different (meaning you can have 1 byte read and write ... and requires software to manage all the communication) from QSPI as it does most of the heavy lifting in the background.

From the Nordic InfoCenter [QSPI](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fps_nrf52840%2Fqspi.html) 
Look at `WRITE.CNT` and `READ.CNT` for example:
[image]

jago85, 2021-08-02:
Thanks for the explanation. Now I understand it. So it's actually an MCU dependent issue. And yes, I assume that most 32 bit systems with a QSPI interface will have this alignment. You could implement a read-modify-write operation for unaligned accesses. But this would lead to bad performance.
In my projects I used STM32 which is an ARM too. But my specific model doesn't have a QSPI interface and I'm only using manually implemented SPI operations.
I might try littlefs sameday too. My main reason would be to have directories. The point that it is under active development is a good reason too.

ShaharHD, 2021-08-02:
> Thanks for the explanation. Now I understand it. So it's actually an MCU dependent issue. And yes, I assume that most 32 bit systems with a QSPI interface will have this alignment. You could implement a read-modify-write operation for unaligned accesses. But this would lead to bad performance.
Correct ... which makes the whole QSPI interface optimization of 4IO (read/write). Further more, most MCU which has QSPI interface, can even use some of the QSPI flash as XIP (execute in place) which makes the useable flash size for code much bigger than the on-board flash.

> In my projects I used STM32 which is an ARM too. But my specific model doesn't have a QSPI interface and I'm only using manually implemented SPI operations.
Some STM do have QSPI interface - it is becoming more and more popular today.

> I might try littlefs sameday too. My main reason would be to have directories. The point that it is under active development is a good reason too.
Worth mentioning that Espressif has their own for SPIFFS in their SDK [Espressif spiffs](https://github.com/espressif/esp-idf/tree/master/components/spiffs) - but I don't think they are compatible and it's designed with FreeRTOS in mind and their ESP32 line of MCUs

peter9477, 2023-10-16:
> You could implement a read-modify-write operation for unaligned accesses. But this would lead to bad performance. 

I thought that as well, and was on the verge of implementing something that did that, until a colleague pointed out this is not necessary.  At least with any flash chip which allows "writing" bytes that are currently erased to a value of 0xFF and thereby leaving them still erased (which I believe is just about any flash chip anyone looking at this would be using, but feel free to correct me if I'm wrong).

If you want to change a single byte within a word (the third one, say) and leave the others alone, don't do this:

1. read current value -> e.g. [0x12, 0xfe, 0xff, 0xff]  // note: last two bytes are erased in this example
2. modify buffer -> [0x12, 0xfe, 0x01, xff]
3. write buffer back to flash, leaving bytes 0, 1, and 3 as they were and changing 2 from 0xff to 0x01.

Instead, just make a buffer with all bytes "erased", then change and write it:

1. make buffer: [0xff, 0xff, 0xff, 0xff]  // no read involved
2. modify your target byte in the buffer: [0xff, 0xff, 0x01, 0xff]
3. write it out...  this will leave bytes 0, 1, and 3 as whatever they were before, since you tried setting their bits all to 0xff, but since you can't change a 0 to a 1, they'll just stay as they were.

Hope that's sufficiently clear...  The flash contents before and after are the same for the two cases, but one of them didn't need to read the contents at all.

igrr, 2023-10-16:
> Worth mentioning that Espressif have their own version of SPIFFS in their SDK Espressif spiffs - but I don't think they are compatible and it's designed with FreeRTOS in mind and their ESP32 line of MCUs

Not really. Espressif SDK uses upstream SPIFFS without any modifications. The files in the directory you have linked are just the porting layer, the adaptation layer to the virtual filesystem framework of the SDK, and the tests.