pellepl/spiffs #265: compiler error

sevstels, 2020-06-04:
Example strings:
    
cfg.hal_read_f = my_spi_read;
cfg.hal_write_f = my_spi_write;
cfg.hal_erase_f = my_spi_erase;

Error: a value of type "void (*)(int, int, char *)" cannot be assigned to an entity of type "spiffs_read"
Error: a value of type "void (*)(int, int, char *)" cannot be assigned to an entity of type "spiffs_write" 
Error: a value of type "void (*)(int, int)" cannot be assigned to an entity of type "spiffs_erase"

jago85, 2020-06-04:
Really hard to guess what the problem is without having a bit more of your code. But from the error message it looks like you don't have the right function prototypes. Just look for spiffs_read how it is defined.

If SPIFFS_HAL_CALLBACK_EXTRA is undefined, it is:
`typedef s32_t (*spiffs_read)(u32_t addr, u32_t size, u8_t *dst);`

Your function seems to return void, but it should look like this:
```
static s32_t my_spiffs_read(u32_t addr, u32_t size, u8_t *dst)
{

    // read the data

    return SPIFFS_OK;
}
```

sevstels, 2020-06-05:
Compiler IAR ARM. I tested all the options listed on the documentation page. The error does not go away. It was possible to fix this only by forced type conversion.

    cfg.hal_read_f = (spiffs_read) my_spiffs_read;
    cfg.hal_write_f = (spiffs_write) my_spiffs_write;
    cfg.hal_erase_f = (spiffs_erase) my_spiffs_erase; 

The compiler probably cannot match types due to the use of macros.

jago85, 2020-06-05:
Casting the funtion pointers is not a good idea. It only hides the problem. Why shouldn't the compile be able to handle macros?
I still think your function prototypes are wrong. But as long as you don't post your code of the three functions, nobody can really help you.

The error message says, your read function looks like this:
`void my_spiffs_read(int addr, int size, char dst)`

There are at least 4 errors:
- Return type must be **s32_t** (not void)
- addr must be **u32_t** (not int)
- size must be **u32_t** (not int)
- dst must be **u8_t*** (not char, your implementation does not even use a pointer here)

Please note that the `my_spi_read` function in the Integration chapter is not the same as `my_spiffs_read`. `my_spiffs_read` calls `my_spi_read`. In your first post you are assigning `my_spi_read` to `cfg.hal_read_f` which will not work if you are following the instructions in the wiki.

sevstels, 2020-06-05:
/*
#define LOG_PAGE_SIZE  256
static spiffs fs;
static u8_t spiffs_work_buf[LOG_PAGE_SIZE*2];
static u8_t spiffs_fds[32*4];
static u8_t spiffs_cache_buf[(LOG_PAGE_SIZE+32)*4];
  
void my_spi_read(int addr, int size, char *buf)
{

}

void my_spi_write(int addr, int size, char *buf)
{

}

void my_spi_erase(int addr, int size)
{

}

static s32_t my_spiffs_read(u32_t addr, u32_t size, u8_t *dst) 
{
    my_spi_read((int)addr, (int)size, (char*)dst);
    return SPIFFS_OK;
}

static s32_t my_spiffs_write(u32_t addr, u32_t size, u8_t *src) 
{
    my_spi_write((int)addr, (int)size, (char*)src);
    return SPIFFS_OK;
}

static s32_t my_spiffs_erase(u32_t addr, u32_t size) 
{
    my_spi_erase((int)addr, (int)size);
    return SPIFFS_OK;
} 

void my_spiffs_mount(void) 
{
    spiffs_config cfg;
    cfg.phys_size = 2*1024*1024; // use all spi flash
    cfg.phys_addr = 0; // start spiffs at start of spi flash
    cfg.phys_erase_block = 65536; // according to datasheet
    cfg.log_block_size = 65536; // let us not complicate things
    cfg.log_page_size = LOG_PAGE_SIZE; // as we said
     
    cfg.hal_read_f = (spiffs_read) my_spiffs_read;
    cfg.hal_write_f = (spiffs_write) my_spiffs_write;
    cfg.hal_erase_f = (spiffs_erase) my_spiffs_erase;    
    
    int res = SPIFFS_mount(&fs,
      &cfg,
      spiffs_work_buf,
      spiffs_fds,
      sizeof(spiffs_fds),
      spiffs_cache_buf,
      sizeof(spiffs_cache_buf),
      0);
//    printf("mount res: %i\n", res);
}
*/

jago85, 2020-06-05:
It is, as I suspected. Your functions do not have the right signatures.

Just change them to exactly match the typedefs in spiffs.h:
```
typedef s32_t (*spiffs_read)(u32_t addr, u32_t size, u8_t *dst);
=> static s32_t my_spiffs_read(u32_t addr, u32_t size, u8_t *dst)

typedef s32_t (*spiffs_write)(u32_t addr, u32_t size, u8_t *src);
=> static s32_t my_spiffs_write(u32_t addr, u32_t size, u8_t *src)

typedef s32_t (*spiffs_erase)(u32_t addr, u32_t size);
=> static s32_t my_spiffs_erase(u32_t addr, u32_t size)
```

**Here is what I can see in your code:**
- `dst` in `my_spiffs_read` must be a pointer
- when calling `my_spi_read` casting `dst` to char is wrong, instead cast to a char pointer
Like this:
```
static s32_t my_spiffs_read(u32_t addr, u32_t size, u8_t * dst)
{
    my_spi_read((int)addr, (int)size, (char *)dst);
    return SPIFFS_OK;
}
```
- `src` in `my_spiffs_write` must be a pointer
- as above, the cast to char is wrong

- `my_spiffs_erase` looks correct
- throw away the function pointer casts

sevstels, 2020-06-06:
Internal functions (my_spi_read etc) are not important, they are just stubs. The main code does not compile even if you delete them.

In any case, thanks for the answers. 
Unfortunately, I have to use LittleFS because your library takes too much RAM.

jago85, 2020-06-06:
It's not my library. Just answering your question. 

Yes, I know that my_spi_read (etc) are stubs. I did not talk about them. The error is in my_spiffs_read (etc) having incorrect parameter types.

In your original post you are using my_spi_read (etc) for the config struct. Which completely matches the compiler error. If you used the corrected versions of my_spiffs_read (etc) the error will disappear.

Don't know if LittleFS takes less memory. I don't have any experience with it.