pellepl/spiffs #266: Unexpected error -10004 when creating&deleting files

MightyPork, 2020-07-01:
Related to #215 and #261 

I build a testbench to evaluate spiffs and other flash filesystems for use with a large (64MB) NOR flash.

After a promising start, I now have a testcase that reliably crashes with code -10004 (some changes also resulted in code -10011). Wait for it to run until the end - it gets slow as the Flash fills up. The crash is in the last loop.

Here's my code:
[spiffs-testbench-crash10004.zip](https://github.com/pellepl/spiffs/files/4859667/spiffs-testbench-crash10004.zip)

What I tried and lead to nothing:
- replace mmapped file with a simple calloc()'d buffer - exact same result
- put volatile where i thought it might have some effect on caching
- turn off SPIFFS caches using the config file

I run this on a regular x86_64 desktop PC with Linux.

I can reproduce it 100% - if you can't, then it might really be some strange machine specific shenanigans with caching. (or something supernatural :ghost:)

pellepl, 2020-07-01:
In spiffs_config.h, did you make sure these types are big enough:

```
// Block index type. Make sure the size of this type can hold
// the highest number of all blocks - i.e. spiffs_file_system_size / log_block_size
typedef u16_t spiffs_block_ix;
// Page index type. Make sure the size of this type can hold
// the highest page number of all pages - i.e. spiffs_file_system_size / log_page_size
typedef u16_t spiffs_page_ix;
// Object id type - most significant bit is reserved for index flag. Make sure the
// size of this type can hold the highest object id on a full system,
// i.e. 2 + (spiffs_file_system_size / (2*log_page_size))*2
typedef u16_t spiffs_obj_id;
// Object span index type. Make sure the size of this type can
// hold the largest possible span index on the system -
// i.e. (spiffs_file_system_size / log_page_size) - 1
typedef u16_t spiffs_span_ix;
```

MightyPork, 2020-07-03:
You are correct, that was indeed the case.

I think `SPIFFS_mount()` could easily catch this - compare the values from `spiffs_config` with `(1ULL << sizeof(spiffs_whatever_ix) * 8ULL) - 1ULL`