pellepl/spiffs #292: SPIFFS_tell returns wrong position after write

4embedded, 2022-11-22:
The following code:
```       
spiffs_file fh = SPIFFS_open(fs, "test.txt", SPIFFS_O_CREAT | SPIFFS_O_APPEND | SPIFFS_O_WRONLY);
s32_t pos = SPIFFS_lseek(fs, fh, 0, SPIFFS_SEEK_END);
cout << pos << endl;
SPIFFS_write(fs, fh, "abcdefghij", 5);
SPIFFS_write(fs, fh, "abcdefghij", 8);
pos = SPIFFS_tell(fs, fh);
cout << pos << endl;
SPIFFS_write(fs, fh, "abcdefghij", 2);
pos = SPIFFS_tell(fs, fh);
cout << pos << endl;
SPIFFS_close(fs, fh);
```
produces the following output (if run 3 times):
```
pos1: 0
pos2: 8
pos3: 15

pos1: 15
pos2: 23
pos3: 30

pos1: 30
pos2: 38
pos3: 45
```

If I change line 1229 in spiffs_hydrogen.c from:
`  res = fd->fdoffset;`
to
`  res = fd->offset;`
I get the following (correct) output:
```
pos1: 0
pos2: 13
pos3: 15

pos1: 15
pos2: 28
pos3: 30

pos1: 30
pos2: 43
pos3: 45
```
Any help or hints about the correct way to fix this would be much appreciated!

Note: we are shipping product with the file system configuration and everything has been working well. I'm confident that the settings are good.

pellepl, 2022-11-22:
Try flushing before ftelling

4embedded, 2022-11-23:
Flushing didn't help. It looks to me like SPIFFS_Tell() already flushes before getting the offset:

```
s32_t SPIFFS_tell(spiffs *fs, spiffs_file fh) {
  SPIFFS_API_DBG("%s "_SPIPRIfd "\n", __func__, fh);
  s32_t res;
  SPIFFS_API_CHECK_CFG(fs);
  SPIFFS_API_CHECK_MOUNT(fs);
  SPIFFS_LOCK(fs);

  fh = SPIFFS_FH_UNOFFS(fs, fh);

  spiffs_fd *fd;
  res = spiffs_fd_get(fs, fh, &fd);
  SPIFFS_API_CHECK_RES_UNLOCK(fs, res);

#if SPIFFS_CACHE_WR
  res = spiffs_fflush_cache(fs, fh);
  SPIFFS_API_CHECK_RES_UNLOCK(fs, res);
#endif

  res = fd->fdoffset;
//  res = fd->offset;  <<-- this seems to work

  SPIFFS_UNLOCK(fs);
  return res;
}
```

closed 2022-11-24 by pellepl

4embedded, 2022-11-27:
Thanks @pellepl ! Works great.

commit 0e3104cd (2022-11-24): fixes #292