pellepl/spiffs #301: Seeking Immediate Guidance: SPIFFS Write Latency Too High on STM32F407

Rielyn31, 2025-05-14:
Description of the Issue:
I am currently facing a performance issue when using SPIFFS on an STM32F407 microcontroller running bare-metal firmware. Specifically, when writing files up to ~4KB, the operation is taking around 1.5 seconds, which is much slower than expected. This high latency is causing significant delays in the application’s performance, and I’m hoping to understand whether this is a known limitation or if there are configuration changes that could help optimize it.

Environment Details:
MCU: STM32F407
Flash: SST26VF080A (256-byte page size)
Filesystem: SPIFFS (port from official GitHub repo)
SPI Driver: STM32 HAL SPI + manual chip-select GPIO
SPIFFS Configuration:
  * Page size: 256 bytes
  * Block size: 256*16 bytes

Questions:
1. Is this write latency expected due to alignment issues with the SST26VF080A flash?
2. Any configuration suggestions for optimizing SPIFFS performance with this flash on STM32F407?

I would greatly appreciate any guidance or insights into improving the performance of SPIFFS on this hardware setup. Thank you in advance for your time and support!

xfma-shang, 2025-05-14:
邮件已收到。自动回复

pellepl, 2025-05-15:
Hi there. It's been like a decade since i looked at this, but iirc 1.5secs sounds awfully long. Unless you're writing to an almost full filesystem needing to GC all the time. Perhaps you can put a printf in the erase callback or something to check this. Also, cache helps of course.

I'd advice you to port to littlefs instead as that is maintained and proven in battle. I have no intentions of diving back into to file system business in foreseeable future. Also, spiffs was originally written for 8k ram CPUs and is extremely restrictive when it comes to memory. At the expense of performance.

Godspeed